gnuradio/gnuradio #3445: Can't use parameter in out of tree blocks

ccskying, 2020-05-02:
I just trying to make a out of tree block and try to define 2 parameters editable in block, but it show like this:

Traceback (most recent call last):
  File "/home/lu/KLD_detector_block.py", line 163, in <module>
    main()
  File "/home/lu/KLD_detector_block.py", line 151, in main
    tb = top_block_cls()
  File "/home/lu/KLD_detector_block.py", line 122, in __init__
    self.howto1_data_generate_three_0 = howto1.data_generate_three()
TypeError: __init__() takes exactly 3 arguments (1 given)

and in the top_block class generated by GNU Radio, I see one line like this:

self.howto1_data_generate_three_0 = howto1.data_generate_three()

my xml is:

  <param>
    <name>Sample size</name>
    <key>sample_size</key>
    <type>int</type>
  </param>
    <param>
    <name>Threshold</name>
    <key>threshold</key>
    <type>float</type>
  </param>

python file of this block:

class data_generate_three(gr.basic_block):
    """
    docstring for block data_generate_three
    """
    def __init__(self, sample_size, threshold):
        gr.basic_block.__init__(self,
            name="data_generate_three",
            in_sig=[numpy.complex64],
            out_sig=[numpy.float32])

velichkov, 2020-05-02:
Hi @ccskying,

You need to add the new parameters in the `<make>` tag as well, something like:
```xml
<make>howto1.data_generate_three($sample_size, $threshold)</make>
```
See also https://wiki.gnuradio.org/index.php/XML_GRC

ccskying, 2020-05-02:
Oh, thank you very much. it works.
it's my fault.

closed 2020-05-02 by ccskying