gnuradio/gnuradio #3446: Remove dependency on Python module `six`

noc0lour, 2020-05-03:
As code on `master` is Python3 only depndency on `six` should be removed and affected code needs to be adapted to use Python3 constructs.

dkozel, 2020-05-03:
There are usages of `six` in the doxyxml code which is potentially tricky to regenerate/port.

GRC is where the majority of imports are and there's probably good starting places there for removing six.

```
$ grep " six$" -r *
docs/doxygen/doxyxml/generated/indexsuper.py:import six
docs/doxygen/doxyxml/generated/compoundsuper.py:import six
gnuradio-runtime/python/gnuradio/eng_notation.py:import six
grc/converter/main.py:import six
grc/gui/Utils.py:import six
grc/gui/Actions.py:import six
grc/gui/ParserErrorsDialog.py:import six
grc/gui/PropsDialog.py:import six
grc/gui/BlockTreeWindow.py:import six
grc/gui/canvas/block.py:import six
grc/gui/canvas/flowgraph.py:import six
grc/core/generator/cpp_hier_block.py:import six
grc/core/generator/hier_block.py:import six
grc/core/platform.py:import six
grc/core/params/param.py:import six
grc/core/schema_checker/utils.py:import six
grc/core/schema_checker/validator.py:import six
grc/core/utils/extract_docs.py:import six
grc/core/utils/init.py:import six
grc/core/utils/expr_utils.py:import six
grc/core/utils/epy_block_io.py:import six
grc/core/utils/descriptors/evaluated.py:import six
grc/core/cache.py:import six
grc/core/blocks/block.py:import six
grc/core/blocks/_flags.py:import six
grc/core/io/yaml.py:import six
gr-digital/python/digital/qa_packet_format.py:import six
gr-digital/python/digital/packet_utils.py:import six
gr-utils/modtool/templates/gr-newmod/docs/doxygen/doxyxml/generated/indexsuper.py:import six
gr-utils/modtool/templates/gr-newmod/docs/doxygen/doxyxml/generated/compoundsuper.py:import six
```

argilo, 2020-10-20:
Re-opening, since several PRs remain to be merged before six is gone.

PR #3799 (2020-10-05): grc: remove six
Partially resolves #3446.

#3467 seems to have stalled and now has merge conflicts, so I thought I'd take a crack at removing six myself beginning with GRC. So far GRC appears to be working normally after these changes.
PR #3811 (2020-10-20): docs: remove six
Partially resolves #3446.

I'm not quite sure how these docs work, but the dependency on six can be easily removed so I've done so here.

The headers suggest these files were auto-generated, but they've been subsequently edited by hand.

Together with #3808 and #3809 the dependency on six will be removed, after which the cmake references can be removed:

https://github.com/gnuradio/gnuradio/blob/4b402404e152ede29cfaa5c0648bdfb5d327038b/CMakeLists.txt#L322
https://github.com/gnuradio/gnuradio/blob/4b402404e152ede29cfaa5c0648bdfb5d327038b/CMakeLists.txt#L347-L352
PR #3809 (2020-10-20): digital: remove six
Partially resolves #3446.

gr-digital still had a few usages of `six.b`. They were only needed because bytes were incorrectly stored in stings. I've updated the code to use bytes where appropriate.
PR #3808 (2020-10-20): runtime: remove six
Partially resolves #3446.

This change removes the last usage of six from the runtime component.
closed 2020-10-20 by mormj