gnuradio/gnuradio #3448: pybind: Unit test fails, just because of comparing tuples to lists fails

marcusmueller, 2020-05-03:
affected tests:
```
FAIL: test_001_fsm (__main__.test_trellis)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/marcus/src/gnuradio/gr-trellis/python/trellis/qa_trellis.py", line 34, in test_001_fsm
    self.assertEqual(fsm_args["awgn1o2_4"],(f.I(),f.S(),f.O(),f.NS(),f.OS()))
AssertionError: Tuples differ: (2, 4, 4, (0, 2, 0, 2, 1, 3, 1, 3), (0, 3, 3, 0, 1, 2, 2, 1)) != (2, 4, 4, [0, 2, 0, 2, 1, 3, 1, 3], [0, 3, 3, 0, 1, 2, 2, 1])

First differing element 3:
(0, 2, 0, 2, 1, 3, 1, 3)
[0, 2, 0, 2, 1, 3, 1, 3]

- (2, 4, 4, (0, 2, 0, 2, 1, 3, 1, 3), (0, 3, 3, 0, 1, 2, 2, 1))
?           ^                      ^  ^                       -

+ (2, 4, 4, [0, 2, 0, 2, 1, 3, 1, 3], [0, 3, 3, 0, 1, 2, 2, 1])
?           ^                      ^  ^                      +

======================================================================
FAIL: test_001_interleaver (__main__.test_trellis)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/marcus/src/gnuradio/gr-trellis/python/trellis/qa_trellis.py", line 57, in test_001_interleaver
    self.assertEqual((K,IN,DIN),(i.K(),i.INTER(),i.DEINTER()))
AssertionError: Tuples differ: (5, (1, 2, 3, 4, 0), (4, 0, 1, 2, 3)) != (5, [1, 2, 3, 4, 0], [4, 0, 1, 2, 3])

First differing element 1:
(1, 2, 3, 4, 0)
[1, 2, 3, 4, 0]

- (5, (1, 2, 3, 4, 0), (4, 0, 1, 2, 3))
?     ^             ^  ^              -

+ (5, [1, 2, 3, 4, 0], [4, 0, 1, 2, 3])
?     ^             ^  ^             +
```
<hr>

```
FAIL: test_002_symbols (__main__.qa_header_payload_demux)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/marcus/src/gnuradio/gr-digital/python/digital/qa_header_payload_demux.py", line 409, in test_002_symbols
    self.assertEqual(header_sink.data(),  header)
AssertionError: [1.0, 2.0, 3.0] != (1, 2, 3)

```

<hr>

marcusmueller, 2020-05-03:
As a solution, I'd propose changing the testing: the returned types seem sensible

mormj, 2020-05-04:
Agree with your solution - I've had to change this in many places and just made the tuples lists.  Not sure why I overlooked trellis

PR #3453 (2020-05-05): trellis: update templated python bindings
also, fix failing qa test

Fixes #3448
closed 2021-03-22 by marcusmueller