gnuradio/gnuradio #3454: pybind: enums in grc cannot be mapped as "int"

mormj, 2020-05-04:
Graphical components that have dropdown enums used to be able to have the .yml as type "int", but after the pybind bindings, these need to be converted to "enum".  This may create issues for using variables to map the type programmatically

branch:pybind

noc0lour, 2020-06-04:
This is missing features is pretty blocking as it prohibits using filters in the flowgraph which require a window type.

mormj, 2020-06-04:
If we change the .yml from int to enum in these blocks that should allow flowgraphs to run.  The only issue is if the window is specified as a variable, the dropdown in grc for enum is then not editable.  Is this what you're referring to?

noc0lour, 2020-06-04:
Is the change from int->enum already on the pybind branch? I just wanted to use a filter, but neither selecting the value in the dropdown nor putting in a "appropriate" int worked.

mormj, 2020-06-04:
Unfortunately it's on a per-block basis, so some have been fixed but not all.  It's inconsistent throughout blocks where the make() calls for enum, some are configured in the yml as int, and some as enum.  Changing the yml from int to enum fixes it on each block.  I will look into seeing if it is possible for the pybinded functions to accept int for enum argument generally

mormj, 2020-06-04:
So I can't get ints to implicitly convert into the enums, but changing the definitions in the yml from "int" to "raw" preserves the dropdown and the ability to modify it with another variable.  Again this is very inconsistent throughout the yml files, so perhaps we should change it throughout?

mormj, 2020-06-04:
I'll go ahead and change the current "int"s to "raw" just so the example files will work

closed 2021-04-28 by mormj