gnuradio/gnuradio #3459: proposal: autogenerate <block>::make implementation

japm48, 2020-05-06:
Apologies if this was already proposed, I tried to search this with no luck.

First of all, I am not sure about the utility of the `make` methods and I would appreciate some pointers on that (no pun intended). I'm guessing it has to do with some "owning" graph structure?; if the underlying reason has to do with SWIG, maybe it would be no longer necessary after Pybind11 migration?

Anyways, the `::make` implementations look like boilerplate to me, unless I'm mistaken, they are always implemented like this (pseudo-C++):
```c++
T::sptr T::make(arg1, arg2, ...)
{
    return gnuradio::get_initial_sptr(
        new T_impl(arg1, arg2, ...));
}
```
IMHO this makes interface changes more error-prone because it requires modifying the list of arguments in yet another place (especially when the list of parameters is big).

So assuming `::make` is still required: is there a possibility of generating this? I'm not suggesting generating C++ source code (that sounds unnecessarily painful), but using templates/macros magic; I'm thinking e.g. about [variadic templates](https://en.wikipedia.org/wiki/Variadic_template) and ["perfect forwarding"](https://eli.thegreenplace.net/2014/perfect-forwarding-and-universal-references-in-c).

Also, if some blocks still require a custom implementation, one can always resort to manual implementation.

noc0lour, 2020-06-18:
In general this sounds like a good suggestion. Although this would mean that 100% of GNU Radio source code will turn into templated `C++` code. 

The construct with make function is coming from the PIMPL approach which was decided a while bac kto be how GNU Radio exposes the public interface of blocks to avoid changing the API too much. Thus each public/exported block interface just calls a `make()` function which is implemented in a `_impl` unit. Changes to the `_impl` block don't propagate to public interface changes and all of your programs will still successfully link to GNU Radio. 

I'm afraid your suggestion is well meaning and I'm a friend of reducing the amount of code myself, but I think it's rather impractical here.

mormj, 2021-04-06:
FYI - for GR 4.0, looking into ways of doing this from one level even higher up:

https://github.com/gnuradio/greps/pull/34

Closing this as a proposal, as I think we have an opportunity to do something more drastic, and incremental improvements here would turn rather painful.

closed 2021-04-06 by mormj