gnuradio/gnuradio #3462: Null pointer deference in gr::blocks::file_meta_source_impl::_open()/tmp/workspace/GNURadio-pipeline/gnuradio/gr-blocks/lib/file_meta_source_impl.cc

markhermeling, 2020-05-07:
Hi,

First report here. Doing static analysis on gnu radio with the CodeSonar static analysis tool.

Found the following in 
gr::blocks::file_meta_source_impl::_open()
in /tmp/workspace/GNURadio-pipeline/gnuradio/gr-blocks/lib/file_meta_source_impl.cc
on line 272.

Starting at line 270:

if (*fp) { // if we've already got a new one open, close it
   fclose(*fp);
   fp = 0;         
}

and then on line 275:
if ((*fp = fdopen(fd, "rb")) == NULL) 

Which would be a Null pointer deference.

I think line 272 should be *fp = 0;

Wondering if I should submit further issues as code changes in git, or as issues. I am happy either way.

Thanks!
Mark

marcusmueller, 2024-01-28:
On failure to open, this would indeed dereference the pointer to address `-1`.

Sorry @markhermeling for letting this lie around so long!

marcusmueller, 2024-01-28:
We're passing a valid pointer to a file pointer to the function (in fact, that's guaranteed: not that the file pointer is valid, but that it has a valid address: it's a member), so both 270 and 275 are fine!

closed 2024-01-28 by marcusmueller