gnuradio/gnuradio #3463: Could NOT find LOG4CPP library

cloudlakecho, 2020-05-08:
This error is during installation.

File: `CMakeLists.txt`
Code line: 
```
# Handle logging
find_package(LOG4CPP REQUIRED)
```

While I run `cmake ../`, I am facing this error:
```
-- Looked for LOG4CPP libraries named log4cpp.
CMake Error at cmake/Modules/FindLOG4CPP.cmake:46 (message):
  Could NOT find LOG4CPP library
```

I checked the log4cpp library in my machine by 
`pkg-config --libs log4cpp` and she looks like having the tool because of the output, `-L/usr/local/lib -llog4cpp`.

I tried to fix this error by myself by doing these:
  (1) I compared two files using `diff /usr/local/src/gnuradio/cmake/Modules/FindLog4cpp.cmake
/usr/local/lib/cmake/gnuradio/FindLog4cpp.cmake` (ref: https://lists.gnu.org/archive/html/discuss-gnuradio/2017-04/msg00024.html)
  (2) I followed code change `log4cpp` -> `LOG4CPP` (ref: https://github.com/gnuradio/gnuradio/pull/2964/commits/b18f9aa7a7db7a5c19844f09b23cb91b74c2f3b3) 
Unfortunately, none of them works... 

Without the code line `find_package(LOG4CPP REQUIRED)`, I could go through installation. 

My environment
    Gnuradio version: 3.8.0.0
    OS: Ubuntu 18.04.4 LTS
    Machine: Nvidia Xavier (CPU: ARMv8 Processor rev 0 (v8l)) 

Please, help me. 

Thanks.

drmpeg, 2020-05-08:
You need both the library and the development package.
```
dpkg -l | grep liblog4cpp5
ii  liblog4cpp5-dev                            1.1.1-3                                          amd64        C++ library for flexible logging (development)
ii  liblog4cpp5v5                              1.1.1-3                                          amd64        C++ library for flexible logging (runtime)
```

cloudlakecho, 2020-05-09:
Thanks, Ron @drmpeg. The installation fixed the LOG4CPP error. I am able to doe `cmake ../` without error.

cloudlakecho, 2020-05-10:
Ron, @drmpeg I realized that I did `cmake ../` at cmake folder not build folder. I still get an error. 

Strangely, if I do in cmake folder there is no error...

japm48, 2020-05-10:
@cloudlakecho cmake tends to mess the directories in unpredictable ways if not run in a separate out-of-source folder. Try cleaning the source folder (e.g. remove gnuradio and uncompress or git clone again) and see if it now works.

If that doesn't work still, is the error message the same as before?

dkozel, 2020-05-10:
@cloudlakecho Depending on where you ran cmake it could be that you have generated build files mixed in with the source directory. japm48's advice is the fastest way, just delete the entire GNU Radio source directory and start over. You can use git's clean feature to remove all the extra files as well if you prefer.

Either way, if you're still having a cmake or compile error then you'll need to paste the full error and log here so we can give advice.

The GitHub issue tracker isn't the best place to work through install issues. You're very welcome to join the chat channel at https://chat.gnuradio.org where a lot of folks are around and can usually help more interactively.

cloudlakecho, 2020-05-13:
Thanks @japm48 and @dkozel for the suggestions on the installation error.

I switched to Pybombs method for Gnuradio installation and it looks like being installed without error.

closed 2020-05-13 by cloudlakecho

no2233t, 2020-07-14:
Hi, I'm  now having the same problem. But I'm a rookie in Linux System and would be grateful if you can show me how I can solve the problem. I mean, the specific instruction. Thx a lot!!

no2233t, 2020-07-14:
@cloudlakecho

cloudlakecho, 2020-07-14:
@no2233t Which Linux OS and version are you using? 

I failed to fix this error (and around three other errors) and I gave up installation using CMake. 

I installed with Pybombs after wiping out everything in my Nvidia Xavier (OS: Ubuntu 18.04). [Here](https://github.com/gnuradio/gnuradio#pybombs) is the instruction and some additional steps in [here](https://github.com/gnuradio/gnuradio/issues/2506#issuecomment-637944199), which I took. I am sorry for not being helpful.

no2233t, 2020-07-15:
@cloudlakecho I'm Using Ubuntu 18.04. I tried using pybombs just now but failed when installing uhd. But the situation is that I have already had a uhd installed beforehand. So, Guess I 'll uninstall all related repositories and start from the very beginning. Thx very much for the info, I'll let you know if things go well, or, I get another problem

no2233t, 2020-07-15:
@cloudlakecho  It seems I've got a problem:
[ 98%] Building CXX object python/CMakeFiles/pyuhd.dir/pyuhd.cpp.o
c++: internal compiler error: 已杀死 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
python/CMakeFiles/pyuhd.dir/build.make:62: recipe for target 'python/CMakeFiles/pyuhd.dir/pyuhd.cpp.o' failed
make[2]: *** [python/CMakeFiles/pyuhd.dir/pyuhd.cpp.o] Error 4
CMakeFiles/Makefile2:4756: recipe for target 'python/CMakeFiles/pyuhd.dir/all' failed
make[1]: *** [python/CMakeFiles/pyuhd.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
Any idea how i can fix this, please?

no2233t, 2020-07-16:
@cloudlakecho  The previous problem is cause by limited RAM space of my VM while building. I've changed my VM settings, everything goes well and I've successfully installed GNU 3.8. Special Thx to you!

mberry28, 2022-11-23:
I am having a similar issue, but with RHEL 8. Can not cmake gnuradio. When i check the log4cpp I produce:
```
 pkg-config --libs log4cpp 
-llog4cpp
```
I have put log4cpp and log4cpp-devel via yum.

Suggestions?

cloudlakecho, 2022-11-29:
@mberry28 I am sorry I floated away from GNURadio. 

I don't think your post receive attention because this post marked as "Closed". Please consider open new "Issue" to receive community attention.