gnuradio/gnuradio #3466: Error in make test Related to qa_codec2_vocode.

cloudlakecho, 2020-05-10:
This error is during installation.

When I try `make test`, it gives an error related to qa_codec2_vocode. It looks like a Python function.
 
Her is my error message:
```
.
.
.
99% tests passed, 1 tests failed out of 346

Total Test time (real) = 321.09 sec

The following tests FAILED:
	333 - qa_codec2_vocoder (Failed)
Errors while running CTest
Makefile:129: recipe for target 'test' failed
make: *** [test] Error 8
```

I searched Internet for the solution, and it looks like I have to install Gunradio using Pybombs...

My environment
Gnuradio version: 3.7.13.4
gcc and g++ version: 7.5.0
cmake version: 3.10.2
OS: Ubuntu 18.04.4 LTS
Machine: Nvidia Xavier (CPU: ARMv8 Processor rev 0 (v8l))

Please, help me.

Thanks.

dkozel, 2020-05-10:
Hi @cloudlakecho Codec2 is one of the voice codecs. If you aren't using that codec then you can ignore this test error.

It is never required to install using PyBOMBS, but it is possible that you have to manually install a more recent version of codec2 than comes by default with Ubuntu 18.04.

If you'd like to debug this test error can you please run
`CTEST_OUTPUT_ON_FAILURE=1 make test`

cloudlakecho, 2020-05-13:
Thanks @dkozel for the suggestions on the installation error. I tried to ignore the error, but I realized that Gnuradio wasn't installed correctly or path wasn't set. 

I switched to Pybombs method for Gnuradio installation and it looks like being installed without error.

closed 2020-05-13 by cloudlakecho