gnuradio/gnuradio #3473: GRC: String type parameters are not evaluated properly in yaml

mrjacobagilbert, 2020-05-12:
Its common to use conditional statements in yaml `make:` templates, for example:

```yaml
make: |-
    module.block(${PARAM1})
    % if PARAM2 == 'string_param':
    self.${id}.set_internal_value(${PARAM2})
    % endif
```

Currently, this will never evaluate as string type parameters are evaluated via `repr()` which results in strings coming back as `"'string_param_val'"` and not `'string_param_val'` which would be expected.

This can either be solved by documentation (use enum types, or use `"'str'"` for comparisons...but where shoudl this go that would be obvious?) or by changing [this line](https://github.com/gnuradio/gnuradio/blob/master/grc/core/params/param.py#L299) to evaluate parameters via `str()` instead of `repr()` though I am not sure of the implications of that.

No preference on my side, haphazardly changing a `repr()` to a `str()` seems dangerous so no PR yet; open to recommendations if an in-tree fix is desired.

drmpeg, 2020-05-12:
I use a lot of conditional statements in the dtv module. Here's a typical make:
```
    make: |-
        dtv.dvb_ldpc_bb(
            ${standard.val},
            % if str(standard) == 'STANDARD_DVBT2':
            ${framesize1.val},
            % else:
            ${framesize2.val},
            % endif
            % if str(standard) == 'STANDARD_DVBT2':
            % if str(framesize1) == 'FECFRAME_NORMAL':
            ${rate1.val},
            % else:
            ${rate2.val},
            % endif
            % else:
            % if str(framesize2) == 'FECFRAME_NORMAL':
            ${rate3.val},
            % elif str(framesize2) == 'FECFRAME_MEDIUM':
            ${rate4.val},
            % else:
            ${rate5.val},
            % endif
            % endif
            ${constellation.val})
```
Looks like str() does the trick. Hope this helps.

mrjacobagilbert, 2020-05-12:
I see the example you provided works fine. We only observe this for out of tree modules, for which even `str(repr('string'))` evaluates to `"'string'"` which != `'string'`.

mrjacobagilbert, 2020-05-12:
One other note: my colleague pointed out this is only true for string type parameters. Enum dtype parameters work just fine. Still something a little unexpected but relatively straightforward to work around in grc yaml.

skoslowski, 2020-05-15:
Note, that these values are not passed into template like that, but wrapped via
https://github.com/gnuradio/gnuradio/blob/master/grc/core/params/template_arg.py

marcusmueller, 2024-01-28:
This might have been affected by my type work in grc/core/ ; need to check

closed 2024-02-14 by marcusmueller