gnuradio/gnuradio #3484: External Libraries in a OOT block of GNU Radio

isaactd92, 2020-05-15:
Hi!
I'm trying to use an external library in GNU Radio. This library is: Intel® Integrated Performance Primitives (Intel® IPP). I use it to create complex arrays type float 32 (Ipp32sc) in a C++ program and separate memory with Malloc (ippsMalloc_32fc). My question is: What are the configurations or steps to add and use an external library , like IPP, in a OOT block?

drmpeg, 2020-05-16:
In lib/CMakeLists.txt, you have to add the libraries to the line "target_link_libraries". Either with -l or if you publish the OOT, you should use find_package() and the macros you created with the find_package() CMake file.

Here's an example from one of my OOT's that links to libpcap and libdvbv5.

https://github.com/drmpeg/gr-ule/blob/master/CMakeLists.txt#L137

https://github.com/drmpeg/gr-ule/blob/master/lib/CMakeLists.txt#L39

https://github.com/drmpeg/gr-ule/blob/master/cmake/Modules/FindPcap.cmake

https://github.com/drmpeg/gr-ule/blob/master/cmake/Modules/FindDvbv5.cmake

noc0lour, 2020-06-18:
This question seems to be better suited for the gnuradio chat https://chat.gnuradio.org or the gnuradio-mailinglist discuss-gnuradio@gnu.org as this bugtracker is reserved to report issues within GNU Radio. Feel free to ask in the chat or mailing-list next time (after looking if someone already had a similar question answered on the mailinglist).

closed 2020-06-18 by noc0lour

isaactd92, 2020-06-19:
Thanks to both !