gnuradio/gnuradio #3487: grc: Background colours in Param Editor are not changed properly

dl1ksv, 2020-05-16:
Changing the type of a value in the grc parameter editor does not change the background colour of the value entry . You have to close and reopen the editor to get the correct background colour in the value entry.

gateship1, 2020-06-09:
Is there more information available for this bug? I have not been able to recreate this issue.
I am assuming this refers to changing the type in the Properties dialog window for a Parameter. When I do this, the background color for the value field is appropriately changing in realtime. Am I not looking at the correct issue?

I am using GNU Radio Companion 3.7.11 on Ubuntu 18.04.4 LTS codename: bionic.

Thank you for your time.

dkozel, 2020-06-09:
@gateship1 I strongly suspect this is a GNU Radio 3.8+ issue. There have been a variety of small display issues over this release cycle.

gateship1, 2020-06-09:
@dkozel, Thank you for the follow-up. I was on GNU Radio 3.8.1 but I switched to the older version because in 3.8.1 there are no colors at all in the Properties dialog window for a Parameter on my machine.
I just rebuilt 3.8.1 but I am still not able to reproduce this issue. Unless the issue is that there is no color (everything is white background in the dialog box).

dkozel, 2020-06-10:
@dl1ksv can you provide any more details? What version of GNU Radio did you encounter this on?

dl1ksv, 2020-06-10:
It's on the master branch  
gnuradio-config-info -v
3.9.0.0-git  

The original type was none.
[image]

Changing the type to float and pressing 'apply' does not change the colour in the value entry.
But if you look at the the attack parameter box in the left upper corner of the flow graph, you see that the type type has already changed to Float. After closing and reopening the background colour will be display correctly.

PR #3935 (2020-12-17): Update parameter background colours realtime
If you change the type of a variable in the Properties Parameter editor the background colour of the parameter value is not updated.
You have to reload the editor for this parameter block to see the background colour according to the parameter type.

Fixes #3487 

This happens because the value box is evaluated before the type box.
Changing the order of the entries to type before value in the parameter.block.yml file fixes this issue.
closed 2020-12-17 by mormj