gnuradio/gnuradio #3488: GRC Opening Error in 3.8.1 Version

cloudlakecho, 2020-05-17:
I have a problem in opening some GRC files. I could open some, but some I couldn't open. 

The Companion crashed without warning message (this is from Terminal):
```
.
.
.
" loaded from
  /home/roman/gnuradio/share/gnuradio/grc/blocks/zeromq_sub_source.block.yml
overwritten by
  /home/roman/gnuradio/share/gnuradio/grc/blocks/zeromq_sub_source.block.yml
<<< Welcome to GNU Radio Companion 3.8.1.0 >>>

Block paths:
	/home/roman/gnuradio/share/gnuradio/grc/blocks
	/home/roman/gnuradio/share/gnuradio/grc/blocks
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
Warning: docstring loader crashed too often

Loading: "/home/roman/Documents/USRP_SinkSource.grc"
>>> Converting from XML
/usr/lib/python3.6/importlib/_bootstrap.py:219: ImportWarning: can't resolve package from __spec__ or __package__, falling back on __name__ and __path__
  return f(*args, **kwds)
>>> Done

Loading: "/home/roman/Documents/CustomBlock.grc"
>>> Converting from XML
roman@jetson:
```

I checked `python3-lxml`, and it is already installed. 

The problematic file saved in Gnuradio verion 3.7.13.4. I found [some information about GRC file format from XML to YAML](https://wiki.gnuradio.org/index.php/GNU_Radio_3.8_OOT_Module_Porting_Guide). The web page suggested using `gr_modtool update --complete` (for my case `pybombs run gr_modtool update --complete` (because I installed Gnuradio using Pybombs), but I get an error:
```
pyboms: error: unrecognized arguments: --complete
```

Do I still need to convert old version GRC file or Gnuradio 3.8 take care the version issue? 

My environment
 Gnuradio version: 3.8.1.0
 gcc and g++ version: 7.5.0
 cmake version: 3.10.2
 OS: Ubuntu 18.04.4 LTS
 Machine: Nvidia Xavier (CPU: ARMv8 Processor rev 0 (v8l))

Please, help me.

Thanks.

ChristopheSeguinot, 2020-05-20:
GNURadio 3.8 automatically convert 3.7 grc files and save them to the new 3.8 format (reverting to 3.7 grc file is not available). I never encountered any problem while opening 3.7 GRC files with GRC 3.8 or 3.9 (excepted for some blocks removed in 3.8 or 3.9). Can you please give an example of 3.7 .grc file you can't open?

cloudlakecho, 2020-05-20:
Hi, @ChristopheSeguinot, I put three GRC files at https://drive.google.com/drive/folders/1PpQ0Gh9QL_D9-LgsAV_bQwOac9XF-Tfd?usp=sharing. 

`CustomBlock.grc` - not able to open
`TxRX_fromFile.grc` - not able to open

`USRP_SinkSource.grc` - able to open

Please, try at version 3.8.1 Gnuradio Companion. Additionally, I installed the Gnuradio with Pybombs, so I run the Gnuradio with `$ pybombs run gnurdio-companion` command. Thanks.

ChristopheSeguinot, 2020-05-20:
At present I only have 3.9.0 installed (compiled from source cause I'm making some development)

- CustomBlock.grc - able to open after python code indentation correction
--This 3.7 GRC file contains an indentation error in the embedded python code. Change indentation on line 495 (remove tab and indent as line 494)
- TxRX_fromFile.grc - able to open (all block connected, no warning, no error)
- USRP_SinkSource.grc - able to open (no connection, only USRP, one sink and one source)

cloudlakecho, 2020-05-21:
@ChristopheSeguinot, thanks for check all three files. It is good to know you could open with change in Python code. I will check the embedded Python code. 

I wonder how to get version 3.9.0? When I check the git, the latest is 3.8.1...Please, check the release tab at https://github.com/gnuradio/gnuradio/releases.

Thanks.

dkozel, 2020-05-26:
@cloudlakecho 3.8.1.0 is the latest stable release. The code on the active development branch `master` will become the 3.9.0.0 release so that is what Christophe is referring to.

You are having a variety of issues while setting up and using GNU Radio. I'd recommend joining the chat channel where you can ask for advice more interactively. The development issue tracker isn't the best place to get this sort of help.
https://chat.gnuradio.org

cloudlakecho, 2020-05-26:
@dkozel Thanks for explanation of 3.9.0 version.
 
Regarding moving the question to Gnuradio chat, I appreciate the suggestion, but I don't want to have another place to follow the specific software issue.

cloudlakecho, 2020-06-03:
@dkozel I decide to join the Chat Chanel and put [my current error about running Gnuradio Companion](https://matrix.to/#/!PiiKkGTcBDLmPnxhoT:gnuradio.org/$gQIk0jz4Rn9Gzp7-rvqJaIeVesqY0M66On21hVpK-J4?via=gnuradio.org&via=matrix.org&via=das-labor.org). I hope I receive some suggestion on this. (I also put the same question in [Git Issue](https://github.com/gnuradio/gnuradio/issues/2506#issuecomment-637809361) as well.

nickoe, 2021-01-10:
@cloudlakecho did you resolve your issue?

cloudlakecho, 2021-01-13:
@nickoe Thanks for checking. Yes, I resolve the issue. My soluation is at [here.](https://github.com/gnuradio/gnuradio/issues/2506#issuecomment-637944199). I don't remember why I didn't close this issue. I close now.

closed 2021-01-13 by cloudlakecho