gnuradio/gnuradio #3494: Gnuradio crashes when trying to use Qt blocks

stevezsuzska, 2020-05-21:
Cent OS 7. Gnuradio 3.8.1.0 Compiled from source.
I have been trying to compile Gnuradio from source. It has been a long journey. Now it finally compiles and opens but when I try to use any Qt instrumentation block or the Qt widget message box, it crashes with a segmentation fault. This appears to be the same problem as described in #3215, where removing the qt-devel package fixed the problem. I removed this package and recompiled but the issue remains. I tried removing other Qt packages such as qt-x11 but qwt 6.1.3 is dependent (6.1.1.1 didn't compile). Here is the gdb output.

[dev@localhost ~]$ gdb -ex run -ex bt --args /usr/bin/python3 /usr/local/bin/gnuradio-companion
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-119.el7
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /usr/bin/python3.6...Reading symbols from /usr/bin/python3.6...(no debugging symbols found)...done.
(no debugging symbols found)...done.
Starting program: /usr/bin/python3 /usr/local/bin/gnuradio-companion
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffe8aed700 (LWP 24437)]
[New Thread 0x7fffe82ec700 (LWP 24438)]
[New Thread 0x7fffe5aeb700 (LWP 24439)]
[New Thread 0x7fffe32ea700 (LWP 24440)]
[New Thread 0x7fffe0ae9700 (LWP 24441)]
[New Thread 0x7fffde2e8700 (LWP 24442)]
[New Thread 0x7fffdbae7700 (LWP 24443)]
[New Thread 0x7fffd92e6700 (LWP 24444)]
[New Thread 0x7fffd6ae5700 (LWP 24445)]
[New Thread 0x7fffd42e4700 (LWP 24446)]
[New Thread 0x7fffcfae3700 (LWP 24447)]
[New Thread 0x7fffbbc48700 (LWP 24449)]
[New Thread 0x7fffbb447700 (LWP 24450)]
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
[New Thread 0x7fffb90e9700 (LWP 24451)]
[Thread 0x7fffe32ea700 (LWP 24440) exited]
[Thread 0x7fffdbae7700 (LWP 24443) exited]
[Thread 0x7fffe5aeb700 (LWP 24439) exited]
[Thread 0x7fffcfae3700 (LWP 24447) exited]
[Thread 0x7fffd6ae5700 (LWP 24445) exited]
[Thread 0x7fffd42e4700 (LWP 24446) exited]
[Thread 0x7fffde2e8700 (LWP 24442) exited]
[Thread 0x7fffe8aed700 (LWP 24437) exited]
[Thread 0x7fffd92e6700 (LWP 24444) exited]
[Thread 0x7fffe82ec700 (LWP 24438) exited]
[Thread 0x7fffe0ae9700 (LWP 24441) exited]
[Detaching after fork from child process 24452]
[New Thread 0x7fffcfae3700 (LWP 24464)]
[New Thread 0x7fffd42e4700 (LWP 24465)]
<<< Welcome to GNU Radio Companion 3.8.1.0 >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks
[Thread 0x7fffd42e4700 (LWP 24465) exited]
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
[Detaching after fork from child process 24469]
Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
[Detaching after fork from child process 24500]
Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
[Detaching after fork from child process 24505]
Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
[Detaching after fork from child process 24507]
Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
Warning: docstring loader crashed too often
[Thread 0x7fffb90e9700 (LWP 24451) exited]

Program received signal SIGSEGV, Segmentation fault.
0x00007fff8e72cb3c in _GLOBAL__sub_I_qprinterinfo.cpp () from /lib64/libQtGui.so.4
#0  0x00007fff8e72cb3c in _GLOBAL__sub_I_qprinterinfo.cpp () at /lib64/libQtGui.so.4
#1  0x00007ffff7dea9b3 in _dl_init_internal () at /lib64/ld-linux-x86-64.so.2
#2  0x00007ffff7def58e in dl_open_worker () at /lib64/ld-linux-x86-64.so.2
#3  0x00007ffff7dea7c4 in _dl_catch_error () at /lib64/ld-linux-x86-64.so.2
#4  0x00007ffff7deeb7b in _dl_open () at /lib64/ld-linux-x86-64.so.2
#5  0x00007ffff7496fab in dlopen_doit () at /lib64/libdl.so.2
#6  0x00007ffff7dea7c4 in _dl_catch_error () at /lib64/ld-linux-x86-64.so.2
#7  0x00007ffff74975ad in _dlerror_run () at /lib64/libdl.so.2
#8  0x00007ffff7497041 in dlopen@@GLIBC_2.2.5 () at /lib64/libdl.so.2
#9  0x00007ffff7a938b6 in _PyImport_FindSharedFuncptr () at /lib64/libpython3.6m.so.1.0
#10 0x00007ffff7a8b965 in _PyImport_LoadDynamicModuleWithSpec () at /lib64/libpython3.6m.so.1.0
#11 0x00007ffff7a8b1f9 in _imp_create_dynamic () at /lib64/libpython3.6m.so.1.0
#12 0x00007ffff799b3c7 in PyCFunction_Call () at /lib64/libpython3.6m.so.1.0
#13 0x00007ffff7a019c3 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#14 0x00007ffff7a05987 in _PyEval_EvalCodeWithName () at /lib64/libpython3.6m.so.1.0
#15 0x00007ffff7a0667a in fast_function () at /lib64/libpython3.6m.so.1.0
#16 0x00007ffff7a06903 in call_function () at /lib64/libpython3.6m.so.1.0
#17 0x00007ffff79fb0f7 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#18 0x00007ffff7a065ca in fast_function () at /lib64/libpython3.6m.so.1.0
#19 0x00007ffff7a06903 in call_function () at /lib64/libpython3.6m.so.1.0
#20 0x00007ffff79fb0f7 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#21 0x00007ffff7a065ca in fast_function () at /lib64/libpython3.6m.so.1.0
#22 0x00007ffff7a06903 in call_function () at /lib64/libpython3.6m.so.1.0
#23 0x00007ffff79fb0f7 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#24 0x00007ffff7a065ca in fast_function () at /lib64/libpython3.6m.so.1.0
#25 0x00007ffff7a06903 in call_function () at /lib64/libpython3.6m.so.1.0
#26 0x00007ffff79fb0f7 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#27 0x00007ffff7a065ca in fast_function () at /lib64/libpython3.6m.so.1.0
#28 0x00007ffff7a06903 in call_function () at /lib64/libpython3.6m.so.1.0
#29 0x00007ffff79fb0f7 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#30 0x00007ffff7a065ca in fast_function () at /lib64/libpython3.6m.so.1.0
#31 0x00007ffff7a06903 in call_function () at /lib64/libpython3.6m.so.1.0
#32 0x00007ffff79fb0f7 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#33 0x00007ffff7a05987 in _PyEval_EvalCodeWithName () at /lib64/libpython3.6m.so.1.0
#34 0x00007ffff7a0667a in fast_function () at /lib64/libpython3.6m.so.1.0
#35 0x00007ffff7a06903 in call_function () at /lib64/libpython3.6m.so.1.0
#36 0x00007ffff79fb0f7 in _PyEval_EvalFrameDefault () at /lib64/libpython3.6m.so.1.0
#37 0x00007ffff7a05987 in _PyEval_EvalCodeWithName () at /lib64/libpython3.6m.so.1.0
Missing separate debuginfos, use: debuginfo-install python3-3.6.8-13.el7.x86_64
---Type <return> to continue, or q <return> to quit---
Any suggestions please?
Steve

velichkov, 2020-05-21:
Hi @stevezsuzska,

> Program received signal SIGSEGV, Segmentation fault.
> 0x00007fff8e72cb3c in _GLOBAL__sub_I_qprinterinfo.cpp () from /lib64/libQtGui.so.4
> #0 0x00007fff8e72cb3c in _GLOBAL__sub_I_qprinterinfo.cpp () at [...]

This crash looks similar to #3215. If you have `qt-devel` package installed try removing it and then execute the gnuradio's tests - in the build directory run `make test`.

stevezsuzska, 2020-05-21:
Hi @velichkov 
Thank you for your reply. I removed qt-devel using yum erase and ran make test. 4 tests fail:

99% tests passed, 4 tests failed out of 364

Total Test time (real) = 181.81 sec

The following tests FAILED:
	 38 - qa_volk_32f_log2_32f (Failed)
	 85 - qa_volk_32fc_s32f_magnitude_16i (Failed)
	348 - qa_qtgui (Failed)
	350 - qa_uhd (Failed)
Errors while running CTest
make: *** [test] Error 8
You have new mail in /var/spool/mail/root

velichkov, 2020-05-21:
>  I removed this package and recompiled but the issue remains. I tried removing other Qt packages such as qt-x11 but qwt 6.1.3 is dependent (6.1.1.1 didn't compile). 

How did you install qwt? Make sure it is compiled with Qt5 and not with Qt4.

> The following tests FAILED:
> 38 - qa_volk_32f_log2_32f (Failed)
> 85 - qa_volk_32fc_s32f_magnitude_16i (Failed)
> 348 - qa_qtgui (Failed)
> 350 - qa_uhd (Failed)

Hmm, lets first try fixing the Qt issue. In https://github.com/gnuradio/gnuradio/issues/3215#issuecomment-592062321 it was reported that a rebuild is not needed but can you try uninstalling gnuradio and doing a clean rebuild - remove the build directory and start from scratch.
```
cd build
sudo make uninstall
mkdir build
cd build
cmake ..
make test
```
If the `qa_qtgui` still fails then run `ctest -R qa_qtgui -V` and attach the full output as file.

stevezsuzska, 2020-05-22:
Thank you again for the reply. I had installed qwt 6.1.3 from an rpm I found because 6.1.1.1 that came with the distribution had a problem. This rpm did indeed appear to have been built on Qt4 so I removed it and built qwt 6.1.4 according to  https://qwt.sourceforge.io/qwtinstall.html. I then deleted the build folder and ran cmake again using -DQWT_INCLUDE_DIRS to point to the include folder on /usr/local/qwt-6.1.4/ and -DQWT_LIBRARIES to point to the libqwt.so file in /usr/local/qwt-6.1.4/lib/. After running make and make test, 3 tests fail:
	 85 - qa_volk_32fc_s32f_magnitude_16i (Failed)
	348 - qa_qtgui (Failed)
	350 - qa_uhd (Failed)
Manually running the qt_gui test:
[root@localhost build]# ctest3 -R qa_qtgui -V
UpdateCTestConfiguration  from :/root/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/root/gnuradio/build/DartConfiguration.tcl
Test project /root/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 348
    Start 348: qa_qtgui

348: Test command: /usr/bin/sh "/root/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh"
348: Test timeout computed to be: 10000000
348: Traceback (most recent call last):
348:   File "/root/gnuradio/gr-qtgui/python/qtgui/qa_qtgui.py", line 24, in <module>
348:     from gnuradio import gr, gr_unittest, qtgui
348:   File "/root/gnuradio/gr-qtgui/python/qtgui/__init__.py", line 38, in <module>
348:     from .range import Range, RangeWidget
348:   File "/root/gnuradio/build/gr-qtgui/python/qtgui/range.py", line 24, in <module>
348:     from PyQt5 import Qt, QtCore, QtWidgets
348: ImportError: /lib64/libQt5Core.so.5: version `Qt_5.13' not found (required by /usr/local/lib64/python3.6/site-packages/PyQt5/QtCore.so)
1/1 Test #348: qa_qtgui .........................***Failed    0.30 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.31 sec

The following tests FAILED:
	348 - qa_qtgui (Failed)
Errors while running CTest

So it's complaining about the version of Qt5. I have v5.9.
When I run gnuradio-companion, it no longer crashes when I try to deploy a Qt instrument, which is an improvement. When I attempt to run the flow diagram, it complains of ImportError: libqwt.so.6 and ModuleNotFoundError : _ptgui_swig . I have added .conf files to /etc/ld.so.conf.d folder to add the qwt library folder on /usr/local/ . I also tried export LD_LIBRARY_PATH....
The qa_volk_32f_x2_pow_32f passes if I run it on its own and qa_uhd fails with a segmentation fault.

stevezsuzska, 2020-05-22:
Embarrassingly, I found the problem leading to the files not being found - I missed the leading / before usr in the .conf file. So so far it's running. Thank you for you help. Should I try to install Qt 5.13? What are the implications of the other 2 failed tests please?

velichkov, 2020-05-22:
Hi @stevezsuzska,

> 348: from PyQt5 import Qt, QtCore, QtWidgets
> 348: ImportError: /lib64/libQt5Core.so.5: version `Qt_5.13' not found (required by /usr/local/lib64/python3.6/site-packages/PyQt5/QtCore.so)

How did you install PyQT5? It seems to be compiled for a newer Qt5 version so you need to uninstall this version an build and install from source.

>  Should I try to install Qt 5.13?

Reinstalling PyQt5 should be easier then updating your Qt5 version.

>  What are the implications of the other 2 failed tests please?

The `volk_32fc_s32f_magnitude_16i` kernel is currently not used in gnuradio but it's interesting why it's failing. You need to report this failure in the [volk](https://github.com/gnuradio/volk/) project - run `ctest -R qa_volk_32fc_s32f_magnitude_16i -V` and provide the full output together with some info about your CPU.

If you have an USRP device you would want `qa_uhd` to work as well, again run `ctest -R qa_uhd -V` and provide the full output.

P.S.
Please use [markdown](https://guides.github.com/features/mastering-markdown/) to format your messages and enclose outputs in triple backticks (```).

noc0lour, 2020-06-18:
Seems like the original issue is solved. If not, please reopen.

closed 2020-06-18 by noc0lour