gnuradio/gnuradio #3495: DVB-T reciprocity issue

therman89, 2020-05-21:
Hi,
I have a recorded IQ stream, which can be demodulated with the dvb-t RX demo just fine. However, when I try to modulate and transmit the resulting MPEG-2 stream, the end result does not correlate with the original IQ stream. I narrowed down the possibilities and I think the "Demod reference signals" - "Reference Signals" blocks might be the problem. If I run the block diagram below, and I do a cross correlation on the input and the output, I see that they are not correlated. However, if I bypass the 2 blocks, I get the same output, as seen on the corresponding figure.

Block diagram with reference signal blocks bypassed:
[image]

Correlation with everything enabled:
[image]

Correlation of input and output when reference signals are skipped:
[image]

It might be possible that I'm doing something wrong or I misunderstand something in the working principle, but in my opinion the input and the output shall correlate with the whole chain enabled.
What am I missing?

drmpeg, 2020-05-21:
It's probably just an alignment issue. The Demod Reference Signals block waits for a sync_start tag from the OFDM Symbol Acquisition block (indicating the beginning of a symbol). Before it receives that tag for the first time, it just discards the samples.

therman89, 2020-05-22:
The correlator does not care about the alignment as it is a cross correlator, so every tick on the X axis is one sample shift. If you look at the X axis of the correlation, that shows that the symmetrical correlator takes 2^24(33,554,432 / 2) samples, which contains about 1638 symbols. I expect that from that many symbols there should be a correlation peak.

drmpeg, 2020-05-22:
If you run the remodulated file back through the receiver flow again, does it decode to the same video/audio?

therman89, 2020-05-22:
Good argument, because that produces seemingly the same MPEG-2 stream, but it is shorter, because of the syncing delay. And if I do this whole round again, so:
- demodulate &rarr; MPEG-2 stream
- modulate &rarr; 1st_remodulated_iq
- demodulate &rarr; bit shorter MPEG-2 stream
- modulate &rarr; 2nd_remodulated_iq

The 2 remodulated IQ streams correlate and the peak is at 2088960 sample offset.
[image]

It is still a mistery to me, why it would not correlate with the original IQ stream, though...
For the record: The IQ file was recorded with a B200 Mini SDR and was resampled to match the DVB-T sampling frequency of 8/7*8MHz.

drmpeg, 2020-05-22:
I think it's just a matter of small changes to the input of an inverse FFT can cause large changes in the output.

dkozel, 2020-05-26:
This is an interesting discussion and it would be educational to see an example showing the effect on the correlation of small changes to the input of the second IFFT (maybe adding some noise?), but I don't think this is a bug or issue with the DVB implementation. I'm going to close the issue but *please* feel very welcome to ask for it to be reopened if you believe there is more investigation needed.

Herman, If you are not already on the chat channel or mailing list I hope you will join. This conversation would be fun to continue and I'd enjoy seeing the results of any further work you do on it.
https://wiki.gnuradio.org/index.php/IRC
https://wiki.gnuradio.org/index.php/MailingLists

closed 2020-05-26 by dkozel