gnuradio/gnuradio #3499: gnuradio-companion: Copy & Paste of block connections does not work

reald, 2020-05-22:
Affected Version: Gnuradio 3.8.1.0 (of Ubuntu 20.04)

# Steps to reproduce:

1. Create flow graph in gnuradio-companion with a few connected blocks
2. Select some blocks and connections between
3. Copy
4. Paste
5. Click 

# Result:
- Only blocks are copied but NOT the connections between them

# Expected result:
- Blocks AND connections are copied

dkozel, 2020-05-26:
@reald This should be fixed in the latest code on the 3.8 branch (`maint-3.8`). If you update and rebuild can you please confirm that the issue is fixed for you? Thanks!

reald, 2020-05-26:
@dkozel  I compiled the maint-3.8 branch in a fresh virtual machine with ubuntu 20.04. I tried and the connections between blocks are still NOT copied. So the bug is NOT fixed in maint-3.8.

```
real@Standard-PC-Q35-ICH9-2009:~/lager/gnuradio$ git rev-parse HEAD
aeba5cc36c0e1445fc017417cbc5b92d2ff91579

real@-Standard-PC-Q35-ICH9-2009:~/lager/gnuradio$ git branch
* maint-3.8
  master

real@-Standard-PC-Q35-ICH9-2009:~/lager/gnuradio$ gnuradio-config-info -v
3.8.1.0
```

skoslowski, 2020-05-26:
AFAIK nothing has been merged yet, to verify the fix one would need to try the version from #3502 (which is for master)

dkozel, 2020-05-26:
Ah! I misinterpreted one of the automated messages above. I'll review that PR in the next day and hopefully merge it and then backport to maint-3.8.

commit dfd31cb1 (2020-05-28): grc: fix pasting connections when block were renamed (fixes #3499)
closed 2020-05-28 by mormj