gnuradio/gnuradio #3506: GRC is broken on maint-3.8 branch

argilo, 2020-05-23:
The change made in #3442 prevents GRC from opening in Python 3 due to a syntax error:
```
$ gnuradio-companion 
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/home/argilo/prefix_38/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/home/argilo/prefix_38/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/home/argilo/prefix_38/lib/python3/dist-packages/gnuradio/grc/main.py", line 73, in main
    from .gui.Platform import Platform
  File "/home/argilo/prefix_38/lib/python3/dist-packages/gnuradio/grc/gui/Platform.py", line 27, in <module>
    from ..core.platform import Platform as CorePlatform
  File "/home/argilo/prefix_38/lib/python3/dist-packages/gnuradio/grc/core/platform.py", line 39, in <module>
    from .generator import Generator
  File "/home/argilo/prefix_38/lib/python3/dist-packages/gnuradio/grc/core/generator/__init__.py", line 19, in <module>
    from .Generator import Generator
  File "/home/argilo/prefix_38/lib/python3/dist-packages/gnuradio/grc/core/generator/Generator.py", line 25, in <module>
    from .hier_block import HierBlockGenerator, QtHierBlockGenerator
  File "/home/argilo/prefix_38/lib/python3/dist-packages/gnuradio/grc/core/generator/hier_block.py", line 7, in <module>
    from .top_block import TopBlockGenerator
  File "/home/argilo/prefix_38/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 18, in <module>
    python_template = Template(filename=PYTHON_TEMPLATE)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 348, in __init__
    module = self._compile_from_file(path, filename)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 430, in _compile_from_file
    code, module = _compile_text(self, data, filename)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 744, in _compile_text
    code = compile(source, cid, "exec")
  File "_home_argilo_prefix_38_lib_python3_dist_packages_gnuradio_grc_core_generator_flow_graph_py_mako", line 42
SyntaxError: from __future__ imports must occur at the beginning of the file
```
@alekhgupta1441

cclamb, 2020-05-23:
I have the same issue on maint-3.8.

alekhgupta1441, 2020-05-23:
@argilo @cclamb I had tested this for python 2.7  but was unaware of this. Should I make a PR to revert this or any syntax correction that you can suggest.

argilo, 2020-05-23:
According to this Stack Overflow post, `from __future__` is not allowed in a Mako template:

https://stackoverflow.com/questions/12537213/using-from-future-import-in-mako-template

It seems like the correct way is to pass a `future_imports` argument into `Template()`:

https://docs.makotemplates.org/en/latest/usage.html

argilo, 2020-05-23:
Actually it seems the intention was to put the `from __future__ import print_function` into the generated flow graph, not execute it in a Mako Python block.

I opened #3508 to fix this.

alekhgupta1441, 2020-05-23:
@argilo Thanks for fixing my fault.

cclamb, 2020-05-23:
I reverted locally. I'm not entirely sure what this was for, but removing it worked for me. But I'm on python 3, so ymmv.

dkozel, 2020-05-24:
Fixed by https://github.com/gnuradio/gnuradio/pull/3508

PR #3508 (2020-05-24): grc: fix broken flow_graph template
Fixes #3506.
Fixes #3141.

The change in #3442 put the `from __future__ import print_function` into a Mako Python block instead of the generated flowgraph. This broke GRC, at least on Python 3.

To fix this I reverted #3442 and moved `from __future__ import print_function` to the correct place. I also put it in a Mako `% if` block so it's only generated in Python 2.
closed 2020-05-24 by dkozel