gnuradio/gnuradio #3507: GRC: Finding Blocks broken (search creates a new flow graph)

reald, 2020-05-23:
Affected version: 3.8.1.0 (Ubuntu 20.04)

# Steps to reproduce:

1. Start gnuradio-companion
2. Create or open flow graph with some blocks
3. Select a block
4. Type "Ctrl + F" for finding blocks. The block is deselected. Type field opens. Type "s" "e" "l" to find selector block. This works as intended.
5. Select a block in the flow graph. The type field is still open and you can continue searching (you can even click in the search field to continue typing). Continue searching by typing "e". -> Works
6. Type "c" => A new flow graph is created

# Expected behaviour:
Typing "c" will just continue search for selector block

# Actual behaviour:
A new flow graph is created. There seems to be a general clash with gnuradio hotkeys. When a block is selected "B" (bypass), "D" (disable), "E" (enable), ... seem to have higher priority than the search field, too.

dkozel, 2020-05-25:
Thanks @reald I've replicated this behavior and agree with you that it's an error in prioritizing/disabling the hotkeys during searching.

argilo, 2020-08-17:
This is indeed an irritating bug, and plenty of people are hitting it:

https://twitter.com/abraxas3d/status/1286379113737891841
https://twitter.com/sharebrained/status/1286397805574070272

A superficial fix was made in https://github.com/gnuradio/gnuradio/pull/2574 but this only handled one of many ways the search box can be reached.

I suspect the problem is related to this code which was commented out during the Python 3 upgrade:

https://github.com/gnuradio/gnuradio/blob/ed972766296683b671f59eca1e83c947f88060f4/grc/gui/DrawingArea.py#L78-L80

That's meant to call this function, which would clear out the block selection when the drawing area loses focus:

https://github.com/gnuradio/gnuradio/blob/ed972766296683b671f59eca1e83c947f88060f4/grc/gui/DrawingArea.py#L213-L218

I was able to restore this code like so:

```diff
diff --git a/grc/gui/DrawingArea.py b/grc/gui/DrawingArea.py
index 5bc9006ea6..03c1152609 100644
--- a/grc/gui/DrawingArea.py
+++ b/grc/gui/DrawingArea.py
@@ -75,9 +75,9 @@ class DrawingArea(Gtk.DrawingArea):
 
         self.connect('leave-notify-event', _handle_notify_event, False)
         self.connect('enter-notify-event', _handle_notify_event, True)
-        # todo: fix
-#        self.set_flags(Gtk.CAN_FOCUS)  # self.set_can_focus(True)
-#        self.connect('focus-out-event', self._handle_focus_lost_event)
+
+        self.set_can_focus(True)
+        self.connect('focus-out-event', self._handle_focus_lost_event)
 
 
     ##########################################################################
@@ -108,7 +108,7 @@ class DrawingArea(Gtk.DrawingArea):
         Forward button click information to the flow graph.
         """
         # The following line causes the canvas to reset position to 0,0
-        #  when changing focus from the console or block search back to 
+        #  when changing focus from the console or block search back to
         #  the canvas
         #  Removing this line leaves the canvas alone when focus changes
         # self.grab_focus()
@@ -212,7 +212,7 @@ class DrawingArea(Gtk.DrawingArea):
 
     def _handle_focus_lost_event(self, widget, event):
         # don't clear selection while context menu is active
-        if not self._flow_graph.context_menu.get_take_focus():
+        if not self._flow_graph.get_context_menu()._menu.get_visible():
             self._flow_graph.unselect()
             self._flow_graph.update_selected()
-            self._flow_graph.queue_draw()
+            self.queue_draw()
```
Unfortunately this doesn't solve the problem, because clicking on another area of the window takes the focus away from the drawing area (and calls `_handle_focus_lost_event` as expected), but clicking back into the drawing area does not restore the focus to it, and so `_handle_focus_lost_event` is not called anymore. I'm not sure why that happens.

argilo, 2020-08-17:
I guess this is the reason why the drawing area doesn't get the focus when clicked:

https://github.com/gnuradio/gnuradio/blob/ed972766296683b671f59eca1e83c947f88060f4/grc/gui/DrawingArea.py#L106-L114

The bug that prompted removing `self.grab_focus()` there can instead be solved by putting the DrawingArea in a Viewport:

```diff
diff --git a/grc/gui/Notebook.py b/grc/gui/Notebook.py
index ef8658a34c..02f27dc316 100644
--- a/grc/gui/Notebook.py
+++ b/grc/gui/Notebook.py
@@ -114,12 +114,15 @@ class Page(Gtk.HBox):
         self.drawing_area = DrawingArea(flow_graph)
         flow_graph.drawing_area = self.drawing_area
 
+        self.viewport = Gtk.Viewport()
+        self.viewport.add(self.drawing_area)
+
         self.scrolled_window = Gtk.ScrolledWindow()
         self.scrolled_window.set_size_request(MIN_WINDOW_WIDTH, MIN_WINDOW_HEIGHT)
         self.scrolled_window.set_policy(Gtk.PolicyType.ALWAYS, Gtk.PolicyType.ALWAYS)
         self.scrolled_window.connect('key-press-event', self._handle_scroll_window_key_press)
 
-        self.scrolled_window.add(self.drawing_area)
+        self.scrolled_window.add(self.viewport)
         self.pack_start(self.scrolled_window, True, True, 0)
         self.show_all()
 
```
With this change, the bug is almost solved. But the "c" and "d" keystrokes are still getting swallowed up in the case where you type something into the search box, select a block, then click back into the search box and type again.

PR #3710 (2020-08-19): grc: prevent search keystrokes from modifying flowgraph 
Fixes #2610.
Fixes #3507.

A confluence of bugs introduced while updating GRC to Python 3 results in keystrokes being sent to the drawing area while typing in the block search box.

First, the drawing area needs to be placed in a Viewport before being added into a ScrolledWindow. Fixing this allows the drawing area to grab focus as intended when it is clicked.

Second, the drawing area needs to unselect all blocks when it loses focus so that the relevant keystroke actions can be disabled. This functionality was commented out during the t [... cut]
closed 2020-08-19 by mbr0wn