gnuradio/gnuradio #3510: GRC: Lack of CI testing makes it too easy to introduce bugs

argilo, 2020-05-24:
As far as I know, the aren't yet any CI tests for GRC. As a result, serious bugs can slip through. A couple recent examples:

https://github.com/gnuradio/gnuradio/pull/3317#issuecomment-629635942
https://github.com/gnuradio/gnuradio/pull/3442#issuecomment-632901964

A couple ideas for some simple tests that might have a good chance of catching bugs:

1. Run `grcc` to compile a basic flow graph, then execute it to verify that it produces the expected output.
2. Iterate over all the `.grc` files in `/*/examples/` folders and compile them with `grcc`, then verify that the output is syntactically valid Python.

mormj, 2020-05-27:
This is an excellent point and a huge weakness in our CI.  There does appear to be some grc related tests that will run when calling ctest, but are not run in the buildbot containers (missing pytest?)

https://github.com/gnuradio/gnuradio/tree/master/grc/tests

But I love the idea of running through all the example flowgraphs in the CI

cclamb, 2020-05-31:
PLEASE to this - I have had two critical, breaking bugs in GRC in the past two months., it's making it very tough to convince folks to use these tools in critical applications.

argilo, 2020-06-02:
One more suggestion (which would have caught #3527):

* Run `grcc` to compile a basic **hierarchical** flow graph, then execute it to verify that it produces the expected output.

marcusmueller, 2024-01-28:
solved in #4208

closed 2024-01-28 by marcusmueller