gnuradio/gnuradio #3512: Gnuradio 3.8: Filter design tool: Cannot set zeros and poles

reald, 2020-05-25:
Affected version: Gnuradio 3.8.1.0 (Ubuntu 20.04)

# Steps to reproduce:
1. Open filter design tool in gnuradio-companion
2. Click on "design"
3. Click "add zero"
4. Click on zero/pole diagram

# Expected behaviour:
Mouse pointer changes to finger over zero/pole diagram. After click a zero is added and filter updates.

# Actual behaviour:
Mouse pointer does not change, no zero added, nothing happens.

# Note:
To make gr filter design tool run it is necessery to change in file /usr/lib/python3/dist-packages/gnuradio/filter/filter_design.py line
   from numpy.fft import fftpack as fft_detail
to
   import numpy.fft as fft_detail

mormj, 2020-05-27:
The interactive features of the filter design tool did not get completely ported over to pyqt5/pygraph, as noted in #2527

marcusmueller, 2024-01-28:
Still present. @reald if you are in the meantime looking for a slightly more polished python filter design tool, try pyfda(x)