gnuradio/gnuradio #3515: Building 3.8.1.0 with Thrift 0.10.0

tinyfx, 2020-05-26:
gnuradio 3.8.1.0
Thrift 0.10.0
Ubuntu 18.04

Building via cmake, I get the following error:

Generating controlport/thrift/gnuradio_types.cpp, controlport/thrift/gnuradio_constants.cpp, controlport/thrift/ControlPort.cpp
[WARNING:/home/john/gnuradio/gnuradio-runtime/lib/controlport/thrift/gnuradio.thrift:47] The "byte" type is a compatibility alias for "i8". Use "i8" to emphasize the signedness of this type.

[FAILURE:generation:1] Error: INVALID TYPE IN type_to_enum: VectorF

gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:61: recipe for target 'gnuradio-runtime/lib/controlport/thrift/gnuradio_types.cpp' failed
make[2]: *** [gnuradio-runtime/lib/controlport/thrift/gnuradio_types.cpp] Error 1
make[2]: *** Deleting file 'gnuradio-runtime/lib/controlport/thrift/gnuradio_types.cpp'
CMakeFiles/Makefile2:851: recipe for target 'gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/all' failed

Any ideas?

It builds ok with Thrift 0.9.2, but that only works with python 2. 

Is there a preferred version of thrift?

Thanks in advance, John

troycurtisjr, 2020-08-16:
Interestingly it does appear to build ok with 3.8.1.0-rc1, which is what the current fedora copr RPM package is using in order to successfully build against thrift 0.10.0 which is included with Fedora 31.

marcusmueller, 2023-12-22:
Mininimum thrift version as of now is at 0.13;  closing this as I can't really produce a system where this build doesn't work. Reopening if we can find one.

closed 2023-12-22 by marcusmueller