gnuradio/gnuradio #3516: Metadata file sink doesn't reflect modified/deleted item tags

eprucka3, 2020-05-26:
@ejk43 

If a tag is modified or deleted from a stream, the tag does not change and continues to be displayed in every following packet in the metadata file sink.

[image]

In this case, a Tags Strobe adds a tag at packet 10. However, the tag is displayed in every packet after the first strobe, as seen in packet 11.

[image]

marcusmueller, 2024-01-28:
can reproduce. Serious bug, as it affects recordings.
[image]

https://gist.github.com/marcusmueller/78dac4f3ae76de87da18d60fc0487c84

marcusmueller, 2024-01-28:
Tagging this "Runtime" and "TSB", because something's fishy here.

marcusmueller, 2024-07-01:
by the way, this happens with inline headers too, and the block only sees the correct tags.