gnuradio/gnuradio #3519: filter: Out of bound memory access in the fft_filter.cc filter() functions

jacobgilbert-snl, 2020-05-28:
Illegal memory access in filter, similar to #3393. It is possible for out of bound memory access within the the fft_filter kernel; in the example below output memory addresses beyond the specified number of items are written to:

```c++
#include <iostream>
#include <vector>
#include <gnuradio/filter/fft_filter.h>

int main()
{
  int ntaps = 5;
  int nsamps = 2000;
  int decimation = 32;

  // generate taps
  std::vector<float> taps(ntaps,1);

  // generate samples
  std::vector<float> samples(nsamps,.5);

  // generate filter
  gr::filter::kernel::fft_filter_fff filt(32,taps,1);

  // output vector - make large enough
  std::vector<float> output((nsamps/decimation)+1, 0.0);

   // filter operates one past length specified
  filt.filter(10,samples.data(),output.data());
  if (output[10] != 0.0) {
    std::cout << "An invalid memory access occurred" << std::endl;
  }
  
  return 0;
}
```

marcusmueller, 2024-01-28:
I hoped #7093 would be the only one of this quality; @argilo  would you take a look?

Repro repo with Makefile for your comfort: https://gist.github.com/marcusmueller/4a16f0faddaf9a52c6fb6bc65e52f0be

```c++
#include <gnuradio/filter/fft_filter.h>
#include <iostream>
#include <vector>

int main() {
  unsigned int ntaps = 5;
  unsigned int nsamps = 2000;
  unsigned int decimation = 32;

  // generate taps
  std::vector<float> taps(ntaps, 1);

  // generate samples
  std::vector<float> samples(nsamps, .5);

  // generate filter
  gr::filter::kernel::fft_filter_fff filt(32, taps, 1);

  // output vector - make large enough
  std::vector<float> output((nsamps / decimation) + 1, 0.0);

  // filter operates one past length specified
  filt.filter(10, samples.data(), output.data());
  if (output[10] != 0.0) {
    std::cerr << "An invalid memory access occurred. Expected 0.0f, got "
              << output[10] << "\n";
  }

  return 0;
}
```

output 

```text
An invalid memory access occurred. Expected 0.0f, got 2.5
```

argilo, 2024-01-29:
From the documentation, it looks like some bounds checks are intentionally omitted:

> Because this is designed as a very low-level kernel operation, it is designed for speed and avoids certain checks in the [filter()](https://www.gnuradio.org/doc/doxygen/classgr_1_1filter_1_1kernel_1_1fft__filter__fff.html#ae60c88dd24246165b98921903c039a31) function itself. The filter function expects that the input signal is a multiple of d_nsamples in the class that's computed internally to be as fast as possible. The function set_taps will return the value of nsamples that can be used externally to check this boundary. Notice that all implementations of the fft_filter GNU Radio blocks (e.g., [gr::filter::fft_filter_fff](https://www.gnuradio.org/doc/doxygen/classgr_1_1filter_1_1fft__filter__fff.html)) use this value of nsamples to compute the value to call [gr::block::set_output_multiple](https://www.gnuradio.org/doc/doxygen/classgr_1_1block.html#a63d67fd758b70c6f2d7b7d4edcec53b3) that ensures the scheduler always passes this block the right number of samples.

In the sample provided, `d_nsamples` is 12.

Although not stated in the documentation, it appears that output samples are also written `d_nsamples` at a time.

marcusmueller, 2024-01-29:
… astonishingly I skipped that paragraph twice while reading the header file.

So, interestingly, if the tap vector was set in the constructor, there's no sane way of getting  `d_nsamples`; there's no accessor. 

- `::ntaps()` returns `d_ntaps` , which is smaller than `d_nsamples`
- `::filtersize()` returns `d_fftsize`, which is `d_nsamples + d_ntaps - 1`; now I'd call that formula an implementation detail

Either you remember the formula from how to get from `filtersize` to `d_nsamples`, or the only way to get `d_nsamples` is to `set_taps(vector<tap_t>)`.

Suspicion: That's a suboptimal API and warrants that someone looks into whether the documentation claim that this is done correctly everywhere. Hm; let me think about that.

argilo, 2024-01-29:
Yes, I would agree that accepting taps in the constructor with no way to fetch `d_nsamples` is a problem. The documentation should probably also be clarified to explain the restrictions on the output buffer.