gnuradio/gnuradio #3520: QT Frequency Sink: Processing of channel stops

reald, 2020-05-28:
# Affected Version: 3.7.x, 3.8.1.0

# Steps to reproduce / actual behaviour:
- Create flowgraph: Noise source, Throttle, IIR Filter, QT GUI Frequency Sink
- QT GUI Range: id=a1, default 0, steps -1...1, stepsize=0.1
- IIR Feed forward taps: 1
- IIR Feedback Taps: [1, a1, 1]
[image]

- Start flowgraph. 
[image]

- Set a1=0.1. Filter gets unstable. FFT signal of this channel disappears completely (no scaling issue, autoscale does not work)
[image]

- Set a1 back to 0. Filter gets stable again (you can verify this with time sink).  FFT Signal does not come back
[image]

# Expected behaviour
- Signal should not disappear or at least come back after filter is stable again.

# Note
- Signal comes back if you change fft size.

mormj, 2021-10-12:
I'm guessing this is due to the inability to plot NaN/Inf values.

marcusmueller, 2024-01-28:
https://gist.github.com/marcusmueller/f020e637f4ea047a7895cbe769219b90

Can reproduce. Added a time sink that shows that values are bounded again after reset of the gain. 
[image]

Note that clicking to hide the filtered signal in the frequency sink and then clicking again to show it does not help, either.