gnuradio/gnuradio #3521: Runtime: Setting block_alias has a severe impact on latency for message blocks

jacobgilbert-snl, 2020-05-28:
Setting the block alias on a message block causes a severe impact on latency. This affects at least maint-3.7, maint-3.8, and master.

This was identified by @jdemel and is easily reproduced if you have a mechanism to validate the time a message is published. The following flowgraph using blocks from gr-timing_utils illustrates this. One set_pdu block has an alias set, the other does not.

[image]

Output after letting it run for a little time:
```
message_counter1 :INFO: Message counter noalias got 4060 messages
message_counter0 :INFO: Message counter alias got 4051 messages
gr::log :INFO: time_delta0 - WALL_CLOCK_TIME_DEBUG (time_delta_noalias): Mean = 0.111014 ms, Var = 0.019549 ms
gr::log :INFO: time_delta1 - WALL_CLOCK_TIME_DEBUG (time_delta_alias): Mean = 125.483784 ms, Var = 5214.583800 ms
```

The distribution of 'latency' values (several are shown below, in ms) is specific and appears to be periodic.

```
time_delta_alias . 241.579
time_delta_alias . 231.052
time_delta_alias . 220.788
time_delta_alias . 210.429
time_delta_alias . 199.37
time_delta_alias . 188.272
time_delta_alias . 177.174
time_delta_alias . 166.546
time_delta_alias . 155.161
time_delta_alias . 144.187
time_delta_alias . 133.57
time_delta_alias . 122.527
time_delta_alias . 111.448
time_delta_alias . 100.6
time_delta_alias . 89.5689
time_delta_alias . 78.373
time_delta_alias . 68.049
time_delta_alias . 56.89
time_delta_alias . 46.1071
time_delta_alias . 34.9669
time_delta_alias . 23.958
time_delta_alias . 12.7881
time_delta_alias . 1.35708
time_delta_alias . 240.599
time_delta_alias . 230.209
time_delta_alias . 219.339
time_delta_alias . 208.341
time_delta_alias . 196.912
time_delta_alias . 185.87
time_delta_alias . 174.934
time_delta_alias . 163.942
time_delta_alias . 152.676
time_delta_alias . 141.518
time_delta_alias . 130.302
time_delta_alias . 119.346
time_delta_alias . 108.357
time_delta_alias . 97.4538
time_delta_alias . 86.7009
time_delta_alias . 76.112
time_delta_alias . 65.033
time_delta_alias . 54.1198
time_delta_alias . 43.175
time_delta_alias . 32.3219
time_delta_alias . 21.6429
time_delta_alias . 10.788
```

It seems that messages are queued up and processed every ~250ms by a means I have been unable to locate. For the path that does not have alias set, the values are consistent and small.

willcode, 2020-05-29:
Not sure what the effects of all this are, but

- basic_block.cc/insert_tail(): global_block_registry.notify_blk(alias())
- block_registry.cc/notify_blk(blk): different paths depending on blk (empty string if no alias)

jdemel, 2020-05-29:
I used a similar setup

Listening on 4042/UDP
PDUSocket -> PDU_block_with_alias_set -> PDUSocket
                                                                      Output to 4043/UDP

Just set or unset the `alias` field in the GRC advanced tab and you'll see the difference. I just copied my Python test code for reference.

```python
import socket
import time

def main():
    rx_address = ('localhost', 4043)
    rx_socket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    rx_socket.bind(rx_address)

    tx_address = ('localhost', 4042)
    tx_socket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

    start = time.time()
    tx_socket.sendto('TEST Payload: lorem ipsum!'.encode(), tx_address)

    data, server = rx_socket.recvfrom(4096)
    end = time.time()
    print(f'From: {server} received {data}, duration: {end - start}')

if __name__ == '__main__':
    main()
```

jdemel, 2020-05-29:
> Not sure what the effects of all this are, but
> 
>     * basic_block.cc/insert_tail(): global_block_registry.notify_blk(alias())
> 
>     * block_registry.cc/notify_blk(blk): different paths depending on blk (empty string if no alias)
This is the function we're looking at.
```cpp
void block_registry::notify_blk(std::string blk)
{
    gr::thread::scoped_lock guard(d_mutex);

    if (primitive_map.find(blk) == primitive_map.end()) {
        return;
    }
    if (primitive_map[blk]->detail().get())
        primitive_map[blk]->detail()->d_tpb.notify_msg();
}
```
I suppose it just returns if `alias()=""` otherwise it'll go down a very different code path which has a terrible effect on latency.

jacobgilbert-snl, 2020-05-29:
This is where our troubleshooting got to as well. If the alias is set, this lookup fails and the thread-per-block notify_msg() function is never called to re-wake up block A. There must be some other mechanism that is periodically waking up blocks which results in emptying the message port backlogs [here ](https://github.com/gnuradio/gnuradio/blob/b8cba61b2b1169004e660ec872fe703f1984880f/gnuradio-runtime/lib/tpb_thread_body.cc#L94). 

It is obvious if you print out the primitive_map that this map only stores the original primitive names, not the aliases (the pdu_set block has the alias):

```
block pdu_set going call notify on this_is_my_alias
could not find block this_is_my_alias in the registry to wake up with notify_msg() 
primitive map is 
message_strobe0 message_strobe
pdu_set0 pdu_set
wall_clock_time0 wall_clock_time
```

In order to fix this, the notify_blk call must use the primitive name (avoid the d_ref_map that stores the alias names - its a pmt dict). However, basic_block does not expose the primitive name. Expose it in basic_block.h and change the notify_blk() call parameters to fix it.

In block.cc it appears that the alias() is used in the primitive_map, but this is deceptive because the alias can be changed at any time. [This primitive register](https://github.com/gnuradio/gnuradio/blob/3d29a9a61a41aea147ec4f1cf21ce198b881fa2b/gnuradio-runtime/lib/block.cc#L50) should instead just use the primitive name.

willcode, 2021-03-03:
Addressed by PR above, closing.

closed 2021-03-03 by willcode