gnuradio/gnuradio #3522: Export Data to Excel

ceeae10, 2020-05-29:
Hi! Is there a way I can export the recorded data of a QT Time Sink to a .csv file or any equivalent? I would like to plot all of the data across the whole execution time and if possible, average all the data? 

Thank you.

mormj, 2020-06-02:
Currently there is no way to export data from a time sink, though this is a good feature request.  For capturing data across the whole execution, I would recommend using file sinks.  Thank you for sharing your use case though, as it does seem like something we should be able to do with the visualization utilities.

ClSlaid, 2021-04-10:
Dear Contributors, I'm new to GR and GRC. I thing it will be nice to have a `format sink` block that can store data into xls, csv or even xml, etc.

While SDR lovers may like to use file sinks to store waves for future replays, a curious data scientist would like to find out which is the most popular time or band for over-air communications, then he/she may need to use GR to collect DFT data from various bands.

Though it's said that users can read from those `file sink`s with `NumPy`'s fromfile() method, storing data to binary is still a little weird and easy to make bugs.

For example, one day you flushed a stream of vector data to file sink, but you accidentally read with complex type but with different size, then everything will possibly go wrong.

However if formatted as csv file, there are less worries, packages like pandas will do dirty jobs for you.

marcusmueller, 2021-05-14:
> For example, one day you flushed a stream of vector data to file sink, but you accidentally read with complex type but with different size, then everything will possibly go wrong.

CSV is really the worst-case of numerical data storage: the truncated floating point format is not only hard to parse efficiently, it's also lossy, and ambiguous (especially in face of internationalization; I've personally had *multiply* occurences where CSV readers and CSV writers didn't agree on the dialect describing decimal separators and number delimiters, as some libraries localize, and some don't). So, if you manage to get reading a raw data file wrong, you'll also manage to get CSV wrong, I'm afraid!

Don't know anyone who'd prefer excel, which is a spreadsheet format with um, it's own very idiosynchratic rules for numerical values, over a data storage format. If your time series data analysis depends on excel, that's fine in general, but for radio data, it's questionable.

Among other well-established data formats, HDF5 is a hot candidate. Since project haystack has a HDF5 exporter, this is already solved!

If you want a well-supported, common RF dataset format, SigMF is the format to go. gr-sigmf might want to see more attention, but for now, I think this is out-of-scope for GNU Radio itself.

closed 2021-05-14 by marcusmueller