gnuradio/gnuradio #3525: OOT's fail to build in OpenEmbedded if gnuradio is built with rm_work

balister, 2020-05-29:
Building an OOT with rm_work enabled leads to this error during configure:

`python3-native/python3
| CMake Error at /home/balister/opensdr/test-builds/dunfell-pi/build/tmp-glibc/work/aarch64-oe-linux/gr-osmosdr/0.2.0-r0/recipe-sysroot/usr/lib/cmake/gnuradio/GrPython.cmake:40 (if):
|   if given arguments:
| 
|     "VERSION_EQUAL" "3"
| 
|   Unknown arguments specified
| Call Stack (most recent call first):
|   /home/balister/opensdr/test-builds/dunfell-pi/build/tmp-glibc/work/aarch64-oe-linux/gr-osmosdr/0.2.0-r0/recipe-sysroot/usr/lib/cmake/gnuradio/GnuradioConfig.cmake:51 (include)
|   CMakeLists.txt:45 (find_package)
| 
| 
| -- Configuring incomplete, errors occurred!
`

balister, 2020-05-29:
The underlyin gissue is when gr-osmosdr is configured, the GrPython cmake looks for the python in the gnuradio build work directory. The rm_work class removes the work directory to save space on the build machine. Disabling rm_work for the gnuradio recipe is a work, but the underlying problem needs fixing.

noc0lour, 2020-06-18:
I wonder if this is caused by L37 in `GnuradioConfig.cmake.in`:
  `set(PYTHON_EXECUTABLE @PYTHON_EXECUTABLE@)` 

Basically for OE you are passing `PYTHON_EXECUTABLE` on the commandline, right? So  we need to have a way, to both, define this in the `GnuradioConfig.cmake` so "regular" users will use the same python version for building OOTs if they don't know what they are doing. And folks that know what they are doing should be still able to pass whatever `PYTHON_EXECUTABLE` pleases them.

mormj, 2021-05-14:
@balister - is this still an issue?

dkozel, 2021-09-29:
@balister can you test if this is still an issue?

mormj, 2022-01-20:
Closing due to lack of response.

closed 2022-01-20 by mormj

jhumphries33, 2022-01-24:
I believe I'm still seeing this or a similar issue on USRP E312/E320. Trying to compile an OOT locally on the USRP with the UHD 4.1.0.5 images.

- [UHD v4.1.0.5](https://github.com/EttusResearch/uhd/releases/tag/v4.1.0.5)

Get the same error as initial post when running the standard build process.

On the E320, able to fix by running cmake like:

`cmake -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_INSTALL_PREFIX=/usr ../`

This compiles and install the new blocks fine.

On the E312, for some reason I also had to edit  /usr/lib/cmake/gnuradio/GnuradioConfig.cmake, line 48 changed to

`set(PYTHON_EXECUTABLE ${PYTHON_EXECUTABLE})`

for it to work correctly (while running the same cmake command as above) based on @noc0lour suggestion above.