gnuradio/gnuradio #3527: flow_graph.py.mako will not generate hier blocks

cclamb, 2020-05-31:
Fresh install via pybombs, version maint-3.8; Error: 

<>---<>
cclamb@muertos:~$ grcc sad_mean_hier.grc 
<<< Welcome to GNU Radio Companion Compiler 3.8.1.0 >>>

Block paths:
	/home/cclamb/.grc_gnuradio
	/home/cclamb/gnuradio/share/gnuradio/grc/blocks

>>> Loading: /home/cclamb/Work/Army/master_dev/gr-bc_spcom/examples/sad_mean_hier.grc
>>> Generating: /home/cclamb/.grc_gnuradio/sad_mean_hierarchical_block.py
>>> Generate Error: /home/cclamb/Work/Army/master_dev/gr-bc_spcom/examples/sad_mean_hier.grc: local variable 'python_version' referenced before assignment
Compilation error
<>---<<>

Current version on master works. Removing references to python_version eliminates the error as well.

cclamb, 2020-05-31:
Seems this commit broke this: 

 d28f4f3c57d4c312f1ab4e7b77d7ceabe557f9fd

removing this commit fixes the problem.

cclamb, 2020-05-31:
Also, removing this line:

% if python_version == 2:
from __future__ import print_function
% endif

and moving the import to the upper section under the first python version check fixes as well.

willcode, 2021-03-03:
Fixed by PR above, closing.

PR #3528 (2020-06-02): grc: always perform version check in flow_graph.py.mako
Just moved the if for hier block down so the python_version is always populated since it is used for both hier and non-hier blocks

Fixes #3527
closed 2021-03-03 by willcode