gnuradio/gnuradio #3532: grc flowgraph generation fails

mtlam, 2020-06-02:
Hi all,

This is very related to Issue #3134, except that the solution there does not workfor me. For reference, I had originally built gnuradio from source on Ubuntu 18.04.4, then had some issues related to getting various additional packages put into place so removed (hopefully all) the various files that were built and installed, and then used `apt` instead to install gnuradio. The error I get is:
>>> Failure
Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/gui/Application.py", line 712, in _handle_action
    generator.write()
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 83, in write
    for filename, data in self._build_python_code_from_template():
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/core/generator/top_block.py", line 122, in _build_python_code_from_template
    'catch_exceptions': fg.get_option('catch_exceptions')
  File "/usr/local/lib/python3/dist-packages/gnuradio/grc/core/FlowGraph.py", line 182, in get_option
    return self.options_block.params[key].get_evaluated()
KeyError: 'catch_exceptions'

This error is nearly identical to the one in the issue above except for the line numbers. I have deleted the cache file and have tried pointing the config file to the correct path for the blocks as per that solution but without success. Indeed, when `grc` loads, it says it is loading from the correct block path. However, another thing that I noticed is that some of the blocks that I want, e.g., those from `gr-osmosdr`, still do not show in `grc` if I search for them. The `.yml` files are in that directory whereas they weren't making it into the one where I built it myself. I don't know if this is related or not.

`gnuradio-config-info -v` return 3.8.1.0. `grc` says it is version 3.9.0.0-git. Any help is appreciated, thanks.

mtlam, 2020-06-03:
For additional reference, I tried an install via pybombs and I have the same issue.

quzn, 2020-06-14:
me too, gr-osmosdr
gnuradio-config-info -v 
3.9.0.0-git

noc0lour, 2020-06-18:
You seem to have multiple conflicting version of GNU Radio intermingeled in `/usr/local` and `/usr`. If you still have the original build directory where you built GNU Radio manually you can go there and do a `sudo make uninstall` to get the files in `/usr/local`. Otherwise you have to manually get rid of `/usr/local/include/gnuradio/*`, `/usr/local/lib64/libgnuradio*`, `/usr/local/lib/python3/dist-packages/gnuradio/*`, `/usr/local/share/gnuradio/*` and probably more places in `/usr/local` which I forgot about. Any additional install through pybombs or manual will end up having multiple installation mixed together and this is a recipe for disaster.

closed 2020-06-18 by noc0lour