gnuradio/gnuradio #3541: Tagged Stream to PDU (and other tagged stream blocks) freezes with large packets

eduardosm, 2020-06-03:
Steps to reproduce:
* Add a Noise Source (or any other source)
* Add a Stream to Tagged Stream block and set a packet length of 10000
* Add Tagged Stream to PDU (its output does not need to be connected to anything)
* Run the flowgraph

The flowgraph will freeze with the following message:
```
/usr/lib/python3.8/site-packages/gnuradio/qtgui/range.py:41: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  precision = len(temp) if temp is not '0' else 0

sched: <block tagged_stream_to_pdu (1)> is requesting more input data
  than we can provide.
  ninput_items_required = 10000
  max_possible_items_available = 8191
  If this is a filter, consider reducing the number of taps.
```

jacobgilbert-snl, 2020-06-03:
This is not going to help with the resolution of this, but tagged stream blocks have quite a number of issues; I would avoid... 

More: #1446

mormj, 2021-05-14:
This can be worked around by setting the minoutbuf on the previous block AND the tagged stream block AND all tagged stream blocks downstream from it

[image]

mbr0wn, 2021-05-14:
Since TSBs have some conceptual issues, the above is the best we can do. Closing.

closed 2021-05-14 by mbr0wn