gnuradio/gnuradio #3551: gr_filter_design does not start

eduardosm, 2020-06-07:
The `gr_filter_design` tool fails to start with the following message:
```
Please install SciPy to run this script (https://www.scipy.org)
```

However, SciPy is actually installed and running `import scipy` from the same Python version works.

Python version: 3.8.3
SciPy version: 1.4.1
OS: Manjaro 64 bits

dkozel, 2020-06-07:
Thanks for the report. The exact trigger for that error is if `from scipy import signal`. Could you please check that just to confirm if it works?
https://github.com/gnuradio/gnuradio/blob/master/gr-filter/python/filter/design/filter_design.py#L38-L41

eduardosm, 2020-06-07:
It seems that it was failing to load fftpack from numpy. I have GNU Radio 3.8.0.0 and this seems to have been fixed in 3.8.1.0, sorry for the false alarm.

closed 2020-06-07 by eduardosm