gnuradio/gnuradio #3567: Python Module open in editor error Win 10 64 bit

khanfar, 2020-06-12:
i install gnu-radio companion under win 10 , when i add python module and try to open in editor , i chose use default , i got error opening an external editor please select different editor ! , i try chose editor , but what should i chose ?? i use win 10 64 bit gnu-radio companion , any help ????

belhadj13, 2020-06-27:
How did you solve the problem, I will be interested to know, 
Thank you for your help !

mohammadjavad78, 2021-01-28:
click on chose editor and try to open with notepad++

mormj, 2022-09-09:
Closing as there is not enough info here to reproduce

SullivanChrisJ, 2022-10-18:
It is easily reproduced. Just create an embedded Python block in Windows and double-click it, and click "Open in Editor". A dialogue to choose the editor presents 3 options 1) Choose an editor, 2) Use the default editor 3) Cancel. Using the default editor gives the error "Unable to load the default editor, please choose an editor". Choosing an editor opens a file dialogue box, which curiously does not allow browsing the entire system. For instance, c:\users\username\AppData is not visible, although a directory called "Application Data" appears in the same place. Navigating into it shows no files, when there are actually hundreds of files there, including the default location of Visual Studio Code. 

I tried setting the default editor to a shortcut to vscode.exe, which could be seen from the dialogue box. That yielded the error "Error opening an external editor. Please select a different editor", after which the clicking the "Open in Editor" button does not open the dialogue to choose the editor, but just produces the same error message.

Restarting gnu radio brought fixed that problem. Then I stumbled upon the right click menu in the file dialogue that adds hidden files to the view, and that works. Absent vscode, perhaps Idle or Notepad would suffice for casual use.

73,
Chris VE3NRT

SullivanChrisJ, 2022-10-19:
Thanks. The instructions [here](https://wiki.gnuradio.org/index.php/Embedded_Python_Block) ("Works in Ubuntu and Debian") do not work in Windows. After restarting GNU radio the chosen editor is forgotten and there is no default, even though double-clicking a .py file brings it up in Visual Studio Code. One must go through the annoying ritual of navigating down 6 levels of directories to select it again.

BlazeRyder, 2022-11-07:
Having the same issue here. I guess it sounds like there is not a work around if you are on windows. When I select choose an editor and then go through the files and select Spyder nothing happens at all. When I click Open in Editor button again after selecting location of Spyder in the file directory nothing happens at all. Nothing opens, no dialogue box........

uchitha1996, 2022-11-20:
i found the error that "app data forlder not showing" because of it is a hidden file. goto properties and unhidden it

RobertSwirsky, 2022-11-21:
It would be nice if setting, for example "editor = " in the config file would work on Windows. Everytime I restart GnuRadio and want to edit an embedded python block, I have to browse around and find "gvim.exe"

pedro-sidra, 2022-12-04:
Hi all, 

This is not a fix, but a hack. Maybe it will help you :)

This is the function that defines what editor will be used (or prompts you to pick one)

https://github.com/gnuradio/gnuradio/blob/b2c9623cbd548bd86250759007b80b61bd4a2a06/grc/gui/Dialogs.py#L405 

I replaced line 405 here with:
```python
def choose_editor(parent,config):
    return r'C:\path\to\vscode.exe'
def _choose_editor(parent, config):
```

For me, Dialogs.py was in: `C:\ProgramData\radioconda\pkgs\gnuradio-grc-3.10.4.0-py310h4a85852_2\Lib\site-packages\gnuradio\grc\gui\Dialogs.py`. This will vary depending on where you installed radioconda and what version you have.

I'm sure there is a more elegant solution, but this worked for me!

rbalciunas, 2023-04-07:
There's a bit of inconsistency in the way config files are used.  In Windows, GRC stores its config values in `C:\Users\Bob\.gnuradio\grc.conf`.  However, the preferred editor value is loaded by the core Gnu Radio framework from `C:\Users\Bob\AppData\Roaming\.gnuradio\config.conf`.  If you open that config file (create it if it doesn't exist) and add the following,
```
[grc]
editor = C:\Program Files\Notepad++\notepad++.exe
```
GRC will automatically launch Notepad++ when you click the "Open In Editor" button. Obviously, change the paths appropriately.

---

The "Use Default Editor" button causes the  `choose_editor()` function in `grc/gui/Dialogs.py` to execute, which returns the name of an executable to use for editing the file.  If one is specified in the config file, the function returns that value.  Otherwise, it tries to return the system-specific executable for launching the default application associated with a filetype (eg. when double-clicking on a .txt file, the system launches the app associated with .txt files).  This executable is called `open` in OSX and `xdg-open` in Linux.  The reason that it doesn't work in this case is because Windows doesn't have an executable for that purpose.  There is a built-in `start` command in the command prompt which does the same thing, but since it's not an executable, it can't be "called" from Python.

A fix might be to change the way the `choose_editor()` dialog works from returning the name of an executable to maybe returning a shell command.  Another way might be to include an executable or batch script or something in an appropriate bin directory, and update `choose_editor()` to return the name of that executable.  Something like
```
process = None
if sys.platform.startswith('linux'):
    process = find_executable('xdg-open')
elif sys.platform.startswith('darwin'):
    process = find_executable('open')
elif sys.platform.startswith('win32'):
    process = find_executable('custom-script-that-launches-file')
if process is None:
    raise ValueError("Can't find default editor executable")
```

Thanks to @pedro-sidra's comment for giving me a hint for where to look.

closed 2023-08-29 by willcode

au5t1nuofa, 2024-04-18:
I am having the same issue but tried everything listed above and I can only open the python code in notepad. Also none of the editors I use are listed in the applications list. Atom, notepad++. Nothing seems to work for me.

[image]

[image]

Hagtronics, 2024-07-01:
I could not find any 'config.conf' file on my system to edit the editor, so...

Here is a 'hack' to force an editor (based on pedro-sidra's comment above),

This is for Windows 10, but should work for any OS as you are bypassing the editor selection (or lack therof) and forcing an editor.

In the Dialogs.py file located at,

C:\Users\steve\radioconda\pkgs\gnuradio-grc-3.10.10.0-py311h329eb19_6\Lib\site-packages\gnuradio\grc\gui\Dialogs.py

(Your exact path might *WILL* different, but this gives you the basic idea of where to look)

I changed this function to a hard-coded selection, (In my case VS Code)
```Python
def choose_editor(parent, config):
    editor = r"C:\Users\steve\AppData\Local\Programs\Microsoft VS Code\Code.exe"
    config.editor = editor
    print(editor)
    return editor
```
Just add a underscore to the existing 'choose_editor()' function - i.e. Make it,
```Python
def _choose_editor(parent, config):
```

Save and you are done.

Open GNU Radio Companion, select a Python Block, edit it and select the editor. Your hard coded solution will now work (the selection will be printed to the python stdout window) and the selection will be remembered between sessions.

(Hopefully!)

marcusmueller, 2024-07-02:
@Hagtronics  this is an unnecessary code modification for something that can be trivially configured; please don't recommend that to others. `gnuradio-config-info --userprefsdir` will tell you where to put your config.

marcusmueller, 2024-07-03:
Thanks! I think this concludes this issue :)