gnuradio/gnuradio #3568: XIO:  fatal IO error 2 (No such file or directory) on X server ":1.0"

tinyxuyan, 2020-06-13:
After I build a very simple flowgraph with gnuradio-companion and execute it, the terminal outputs the messages as below:

```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-xuyan'
The X11 connection broke: Maximum allowed requested length exceeded (code 4)
XIO:  fatal IO error 2 (No such file or directory) on X server ":1.0"
      after 447 requests (443 known processed) with 0 events remaining.
```
In my idea, there should be a figure about the signal in time domain. However, there is nothing displaying.

tinyxuyan, 2020-06-15:
ENV: Ubuntu 18.04 LTS; Tigervnc
I guess the reason is that I am using gnuradio in remote access using tigervnc.

dkozel, 2020-06-15:
You're correct, the issue is that the graphics manager cannot find an X Server to connect to.

TigerVNC has an X Server but it wasn't immediately clear to me what extra steps are needed to make it work. I recommend finding a support forum for TigerVNC and asking there.

If your local computer is also Linux you can use `ssh -X USERNAME@REMOTE_HOSTNAME` to be able to run graphical commands like GNU Radio Companion.

This isn't an issue with GNU Radio, so I'm going to close the issue. I hope you're able to get the connection working.

closed 2020-06-15 by dkozel

generatz, 2021-03-02:
FWIW, I had a 
`Gdk-Message: 11:11:10.047: wireshark: Fatal IO error 2 (No such file or directory) on X server :55` (I was using Xvnc on port 5955).
I discovered the problem to be a leftover LD_LIBRARY_PATH set in the bash session I was calling wireshark from. I just needed to remove it.