gnuradio/gnuradio #3569: runtime/blocks (?): tag propagation anomalies when selectively dropping data

mrjacobagilbert, 2020-06-13:
While debugging a simple block, we noticed some unusual tag propagation behavior when dropping large amounts of data. This can be demonstrated using the in-tree selector block as follows:

[image]

Initially, the tags are correct:

[image]

However, by selecting and deselecting the 'enable' check box a couple times, its clear the tag propagation has some errors:

[image]

We attributed this to a runtime issue, as it appears that runtime appears to have sufficient information to propagate tags correctly (consuming X samples, producing 0 samples), though this is a bit of a special case of `general_work()` and propagation can also be fixed in the block directly via manual tag propagation in the selector block.

Regardless, tags should propagate correctly through this block. Thoughts on the correct approach?

jacobgilbert-snl, 2020-06-14:
I now think runtime is behaving as expected, and tag offsets need to be modified manually based on how many samples have been dropped on each output stream.

This gets more complicated if the input streams can be at different rates, however it does not appear this block will function under those conditions.

I’ll get a PR together.

rear1019, 2020-06-15:
This reminds me of an issue concerning tag propagation I have been hit with twice some time ago. I remember tracking it down to a (IMO) questionable behavior in tag propagation, but I never got around to reporting the error… The code in `selector_impl.cc` is probably affected: It boils down to the fact that tags are forwarded even if the following conditions apply:
* `general_work()` consumes *all* input items
* `general_work()` produces *no* output items, i.e. either `return 0` or `produce(output_index, 0) + return  WORK_CALLED_PRODUCE` is executed.

These conditions can occur in the selector block.

I’ll try to come up with code to test this.

rear1019, 2020-06-15:
Follow-up: I should have read the first comment more carefully:
> […] its possible to see that old tags remain in the output buffer with newer data (runtime does not account for the updated tag placement in the output buffer now that some data has been dropped?).

Yes, that’s exactly what happens. 

> We attributed this to a runtime issue, as it appears that runtime appears to have sufficient information to propagate tags correctly (consuming X samples, producing 0 samples), though this is a bit [...]

Manual tag propagation (accounting for dropped samples) does solve the issue. However, this is a runtime issue (in my opinion).

mrjacobagilbert, 2020-06-16:
For now, this has been resolved by manual tag propagation with offsets adjusted to compensate for dropped data.

I go back and forth on whether or not runtime can solve this effectively for general blocks... this is a specific case so its easy to define the desired behavior here but I am not sure in general. Either way TPP_ALL_TO_ALL makes no sense for a block like the selector.

PR #3712 (2020-09-09): blocks: align output tags in selector block
Aligns the output tags using nitems_written as recommended by @rear1019 in #3572 to address the situation where tags on the input are not aligned with the output due to dropped samples

@mrjacobagilbert @rear1019 - please take a look, I believe this can supercede #3572 with less bookkeeping

Fixes #3569
closed 2020-09-09 by mormj