gnuradio/gnuradio #3571: Float to Char behaviour differs between arm and x64

matburnham, 2020-06-15:
`Float to Char` block behaviour appears to be suboptimal on a Raspberry Pi when compared to my laptop. Negative values appear to be truncated to zero on arm but not on x86_64.

### Steps to reproduce

Create a simple GRC flowgraph (or use [attached](https://github.com/gnuradio/gnuradio/files/4780636/bug.zip)):

[image]

1. Run on x86_64:

```
mat@devbox:~/satnogs/bug$ ./test.py
mat@devbox:~/satnogs/bug$ hexdump test.out
0000000 407f 0000 81c0                         
0000006
```

2. Run on arm:

```
pi@pinog:~/test $ ./test.py
pi@pinog:~/test $ hexdump test.out
0000000 407f 0000 0000
0000006
```

3. Spot the difference.
4. Profit.

### Workaround

The following bodge appears to work around the issue:

[image]

See https://community.libre.space/t/demodulate-lrpt-from-satnogs-iq-file/6167/56 for further workings. I do wonder if this relates to #2022.

rear1019, 2020-06-15:
This is an error in VOLK as `Float To Char` simply calls `volk_32f_s32f_convert_8i()`. Please create a bug report in [1].

[1] https://github.com/gnuradio/volk

mormj, 2021-04-06:
Closing since this is a volk issue and appears to have been fixed there.

closed 2021-04-06 by mormj