gnuradio/gnuradio #3573: Windows GRC 3.8 start-up fail

MountainLogic, 2020-06-16:
I have a recent WIN 10 image, installed 3.7 then Visual Code's python packages, pothos, then 3.8 and then 3.8 crashed on start.  I used the 64-Bit Any CPU versoins from http://www.gcndevelopment.com/gnuradio/downloads.htm. I uninstalled 3.7, 3.8 and pathos and installed 3.8 and now I things hang on start.  Messages when started from a command box:

`
C:\Program Files\GNURadio-3.8\bin>"C:\Program Files\GNURadio-3.8\bin\run_gr.bat" gnuradio-companion.py                                                                                               setting gnuradio environment                                                                                                                                                                         WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)                                                                                                                         Traceback (most recent call last):                                                                                                                                                                     File "gnuradio-companion.py", line 102, in <module>                                                                                                                                                    run_main()                                                                                                                                                                                         File "gnuradio-companion.py", line 95, in run_main                                                                                                                                                     exit(main())                                                                                                                                                                                       File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\main.py", line 83, in main                                                                                                          platform.build_library()                                                                                                                                                                           File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\core\platform.py", line 197, in build_library                                                                                       utils.hide_bokeh_gui_options_if_not_installed(self.blocks['options'])                                                                                                                              File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\core\utils\backports\chainmap.py", line 36, in __getitem__                                                                          return self.__missing__(key)            # support subclasses that define __missing__                                                                                                               File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\core\utils\backports\chainmap.py", line 28, in __missing__                                                                          raise KeyError(key)                                                                                                                                                                              KeyError: 'options'                        

'               

Also, if I open a clean dos box I can run:

'
C:\Program Files\GNURadio-3.8\bin>run_gqrx.bat
setting gnuradio environment

C:\Program Files\GNURadio-3.8\bin>

' 
but gqrx does not start

MountainLogic, 2020-06-16:
Also my path included the following:
C:\Program Files (x86)\Python38-32\Scripts\;C:\Program Files (x86)\Python38-32\;;C:\python27;C:\python27\Scripts;

dkozel, 2020-06-17:
@MountainLogic The `options` key error is caused by a bad setting being carried from 3.7 to 3.8 in the configuration file. This has been fixed in the 3.8.1.0 release but there aren't updated installers available from GNC Development yet.

If you can delete the configuration file that is expected to clear the error and allow GRC to launch normally. You can find the preferences(configuration) directory using the following command. I'm not sure where you'll need to run it, it needs to be in the environment that includes the installed GNU Radio binaries.

```
$ gnuradio-config-info --userprefsdir
/home/dkozel/.gnuradio
```

MountainLogic, 2020-06-17:
Thanks so much @dkozel, I appreciate the quick and detailed response .  That resolved the issue like a charm.  

For the next person, from a dos command line window, I did the following (running and failing to start GRC is a required step to set the environment).

C:\Program Files\GNURadio-3.8\bin>run_grc                                                                                                                                                                                                                                                                                     C:\Program Files\GNURadio-3.8\bin>CALL run_gr.bat gnuradio-companion.py                                                                                        setting gnuradio environment                                                                                                                                   WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)                                                                                   Traceback (most recent call last):                                                                                                                               File "gnuradio-companion.py", line 102, in <module>                                                                                                              run_main()                                                                                                                                                   File "gnuradio-companion.py", line 95, in run_main                                                                                                               exit(main())                                                                                                                                                 File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\main.py", line 83, in main                                                                    platform.build_library()                                                                                                                                     File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\core\platform.py", line 197, in build_library                                                 utils.hide_bokeh_gui_options_if_not_installed(self.blocks['options'])                                                                                        File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\core\utils\backports\chainmap.py", line 36, in __getitem__                                    return self.__missing__(key)            # support subclasses that define __missing__                                                                         File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\grc\core\utils\backports\chainmap.py", line 28, in __missing__                                    raise KeyError(key)                                                                                                                                        KeyError: 'options'                                                                                                                                                                                                                                                                                                           C:\Program Files\GNURadio-3.8\bin>gnuradio-config-info.exe --userprefsdir                                                                                      C:\Users\ML\AppData\Roaming\.gnuradio      

now go to where the gnuradio-config-info.exe result pointed and delete the config file.  I left the pref directory & contents in place and things seem to start ok.

closed 2020-06-17 by MountainLogic