gnuradio/gnuradio #3576: No gr-osmosdr blocks visible in companion

mtlam, 2020-06-19:
Hello,

This is related to the second half of #3532. My `gnuradio-companion` is no longer crashing but I still am unable to find any of the osmosdr blocks in the GUI. My `~/.gnuradio/config.conf` `global_blocks_path` is pointing at the correct directory where I see the two osmosdr yml files. I have removed `~/.cache/grc_gnuradio/cache.json` as well with no success. Am I supposed to be pointing to these files somehow else? Thanks.

willcode, 2020-10-28:
Likely configuration problem, and probably solved by now. @mtlam can this be closed?

mtlam, 2020-10-28:
@willcode I never got a solution to this and also hadn't gotten a response on Matrix either, so I had to try to make some demonstrations without gnuradio. I would still like to use it if possible given all of the capabilities, though.

willcode, 2020-10-28:
In https://github.com/gnuradio/gnuradio/issues/3532 you mention grc 3.9, along with a gnuradio 3.8 install. Mixed versions will cause problems. gr-osmosdr and gnuradio 3.9 don't get along quite yet (they should soon). I'm not sure what files you need to delete or env vars you need to change, but try installing 3.8 in a clean environment. If you are building from source, use the maint-3.8 branch.

closed 2020-10-28 by willcode

mtlam, 2020-10-28:
Ah, yes, I tried this prior to heading over to Matrix. I've tried clean installs for multiple versions via multiple methods (source, PyBOMBS, apt) and never had success; and I did verify that all of the files were removed. I can try again though am not hopeful, and have mostly been waiting for improved connectivity between the two like you mentioned.