gnuradio/gnuradio #3577: Replace wavfile_sink/_source with more capable alternative

marcusmueller, 2020-06-19:
WAV is actually just a container, and the types of contained formats that our wavfile source reads is pretty restricted.

I'd propose a libsndfile-based audio file source and sink. Since 2009, libsndfile can also read and write FLAC and Vorbis files, and the ability to work with compressed audio is a frequenty requested feature.

As benign a dependency libsndfile is, it's still a new dependency.

If we add new blocks, we should deprecate the current wavfile blocks.

drmpeg, 2020-06-20:
I have the source block working (with 24-bit 96K FLAC) already.

marcusmueller, 2020-06-20:
> I have the source block working (with 24-bit 96K FLAC) already.

Hah! Awesome!

closed 2020-09-20 by drmpeg