gnuradio/gnuradio #3581: Investigate "The package name passed to `find_package_handle_standard_args` (PkgConfig) does not match the name of the calling package"

marcusmueller, 2020-06-19:
```
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Codec2).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:45 (find_package_handle_standard_args)
  cmake/Modules/FindCodec2.cmake:12 (INCLUDE)
  gr-vocoder/CMakeLists.txt:36 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

Happens for Codec2, 
FFTW3f, 
GMP, 
GSL, 
GSM, 
MPIR, 
PORTAUDIO, 
THRIFT, 
UHD, 
Volk, 
ZeroMQ.

CMake 3.17.0, Fedora 30

drmpeg, 2020-06-20:
This can be reproduced on Ubuntu by just using CMake 3.17 or later.

See also https://github.com/gnuradio/volk/issues/387

nslogan, 2021-05-15:
### Proposed Fix

I believe this pattern of lines, for example:

https://github.com/gnuradio/gnuradio/blob/6dbead201670462cfade299da6c0982e55663678/cmake/Modules/FindCodec2.cmake#L12-L14

Should actually be:

```cmake
if(NOT PKG_CONFIG_FOUND)
  include(CMakeFindDependencyMacro)
  find_dependency(PkgConfig)
endif()
```

I use [`find_dependency`](https://cmake.org/cmake/help/latest/module/CMakeFindDependencyMacro.html) instead of `find_package` because this is exactly what it's designed to do:

> It is designed to be used in a Package Configuration File (`<PackageName>Config.cmake`)

### Root Cause

The line `INCLUDE(FindPkgConfig)` triggers a call within in `FindPkgConfig.cmake` to `find_package_handle_standard_args` with a `<PackageName>` of `PkgConfig`; however, the call to `find_package` for, e.g. `Codec2`, has already set `CMAKE_FIND_PACKAGE_NAME` to, e.g. `Codec2`. This triggers a check in `FindPackageHandleStandardArgs.cmake` which was added in 3.17:

```cmake
  if (DEFINED CMAKE_FIND_PACKAGE_NAME
      AND NOT FPHSA_NAME_MISMATCHED
      AND NOT _NAME STREQUAL CMAKE_FIND_PACKAGE_NAME)
    message(AUTHOR_WARNING
      "The package name passed to `find_package_handle_standard_args` "
      "(${_NAME}) does not match the name of the calling package "
      "(${CMAKE_FIND_PACKAGE_NAME}). This can lead to problems in calling "
      "code that expects `find_package` result variables (e.g., `_FOUND`) "
      "to follow a certain pattern.")
  endif ()
```

There is [an issue](https://gitlab.kitware.com/cmake/cmake/-/issues/20549) that was opened and then closed without any update in CMake's GitLab Issues, I suspect the author of that issue discovered the same source of the error but never posted an update :man_shrugging:.

drmpeg, 2021-05-15:
@nslogan This issue has already been fixed by the pull requests mentioned above, but we neglected to close it.

closed 2021-05-15 by drmpeg

nslogan, 2021-05-15:
Oh, good to know @drmpeg, thanks!

Just for posterity, in case there's some refactoring in the future, the issue actually still exists but has been masked by deferring the include unless `PKG_CONFIG_FOUND` isn't defined. Because `include(FindPkgConfig)` is included in the top-level CMakeLists.txt (or, really, as long as it is included outside of a Find Package file), the IF statement will never fire.

However, I did a quick test and added a `find_package(Codec2)` call _before_ the `include(FindPkgConfig)` in the top-level CMakeLists.txt and the issue crops back up:

```
CMake Warning (dev) at /usr/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Codec2).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.20/Modules/FindPkgConfig.cmake:70 (find_package_handle_standard_args)
  cmake/Modules/FindCodec2.cmake:13 (INCLUDE)
  CMakeLists.txt:26 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

As long as PkgConfig has been located _before_ any call to the Find Packages using the if-include signature in the PR then it will be fine. Otherwise, the fix I posted above will work :+1:.