gnuradio/gnuradio #3583: pybind: bind_gr_module script should not depend on installed prefix

mormj, 2020-06-20:
Currently one of the parameters for generating bindings on an in-tree module is prefix, e.g:

```
python3 /share/gnuradio/grdev/src/gnuradio/gr-utils/bindtool/scripts/bind_gr_module.py --prefix /share/gnuradio/grdev --output_dir /share/tmp/test dtv
```

where prefix points to where gr is installed.  Prefix is only used to point pygccxml to the header files referenced by the one that is being binded.  This really should be done entirely in the src directory.

mormj, 2020-06-22:
Theoretically, the changes to not rely on prefix are pretty simple, however there is one issue in that config.h doesn't live in the source tree.  Unfortunately pygccxml compiles everything needed by the header file that is attempting to be binded, so it needs all the corresponding header files.  Just using the source include dirs leads to an error like this:
```
python3 /share/gnuradio/grdev/src/gnuradio/gr-utils/bindtool/scripts/bind_gr_module.py --output_dir /share/tmp/test blocks
/share/gnuradio/grdev/src/gnuradio/gr-utils/bindtool/scripts
Namespace(include='', names=['blocks'], output_dir='/share/tmp/test', prefix=None, src='/share/gnuradio/grdev/src/gnuradio/gr-utils/bindtool/scripts/../../..')
INFO Parsing source file "/share/gnuradio/grdev/src/gnuradio/gr-blocks/include/gnuradio/blocks/abs_blk.h" ... 
In file included from /share/gnuradio/grdev/src/gnuradio/gr-blocks/include/gnuradio/blocks/abs_blk.h:16:
In file included from /share/gnuradio/grdev/src/gnuradio/gnuradio-runtime/include/gnuradio/sync_block.h:15:
In file included from /share/gnuradio/grdev/src/gnuradio/gnuradio-runtime/include/gnuradio/block.h:15:
In file included from /share/gnuradio/grdev/src/gnuradio/gnuradio-runtime/include/gnuradio/basic_block.h:27:
In file included from /share/gnuradio/grdev/src/gnuradio/gnuradio-runtime/include/gnuradio/rpcregisterhelpers.h:17:
/share/gnuradio/grdev/src/gnuradio/gnuradio-runtime/include/gnuradio/rpcserver_selector.h:14:10: fatal error: 
      'gnuradio/config.h' file not found
#include <gnuradio/config.h>
         ^~~~~~~~~~~~~~~~~~~
```

Further reason to try and get rid of pygccxml

marcusmueller, 2024-02-14:
gnuradio/config.h is included in install set now.

closed 2024-02-14 by marcusmueller