gnuradio/gnuradio #3584: get rid of `from __future__ import stuff`

marcusmueller, 2020-06-20:
`unicode_literals` was necessary pre-Python 2.6; `absolute_import` pre-2.5.

It's especially not necessary on Py3.

Hence in the sence of cleaning up the source tree, deletion of all `from __future__ import unicode_literals` from the python files is desirable. This should ideally not happen manually, but with a script, or command line: we need to do that on `master`, but it might also be desirable to offer tooling to OOT developers.

Reference for the `__future__` module: https://docs.python.org/3.6/library/__future__.html#module-__future__

85ayush, 2021-02-23:
I would love to work on this issue, I am new to open source contribution, can you please guide me?

marcusmueller, 2021-03-22:
In this case, 85ayush, you would have had to find all uses of the mentioned construct and replace them with the appropriate modern construct.

Luckily, this has happened already, and this issue has gone stale.

closed 2021-03-22 by marcusmueller