gnuradio/gnuradio #3595: pybind11: pygccxml deprecation issues

mormj, 2020-06-22:
For generating python bindings beyond the `make()` function, currently we require pygccxml to parse the header file, and then some in-tree scripts to take the output and convert that to the binding code.

The latest version of pygccxml (1.9.1) in pypl is quite a bit behind the develop branch, and has some deprecation issues with python 3.9, namely `module 'time' has no attribute 'clock'`

This leaves 2 options
1) Install pygccxml using setup.py instead of pip
2) Get rid of pygccxml as a dependency

drmpeg, 2020-06-22:
Even the develop branch of pygccxml doesn't seem to be working correctly. It's not creating good .cc files. Here's an example (divide_python.cc):
```
*
 * Copyright 2020 Free Software Foundation, Inc.
 *
 * This file is part of GNU Radio
 *
 * SPDX-License-Identifier: GPL-3.0-or-later
 *
 */

/***********************************************************************************/
/* This file is automatically generated using bindtool and can be manually edited  */
/* The following lines can be configured to regenerate this file during cmake      */
/* If manual edits are made, the following tags should be modified accordingly.    */
/* BINDTOOL_GEN_AUTOMATIC(0)                                                       */
/* BINDTOOL_USE_PYGCCXML(0)                                                        */
/* BINDTOOL_HEADER_FILE(divide.h)                                        */
/* BINDTOOL_HEADER_FILE_HASH(aba6049c528454ca838f70d73f072609)                     */
/***********************************************************************************/

#include <pybind11/complex.h>
#include <pybind11/pybind11.h>
#include <pybind11/stl.h>

namespace py = pybind11;

#include <gnuradio/blocks/divide.h>
// pydoc.h is automatically generated in the build directory
#include <divide_pydoc.h>

void bind_divide(py::module& m)
{

}

```

Oddly, the hash matches.

mormj, 2020-06-23:
Yes, templated blocks are particularly challenging for pygccxml and I did all those manually :(

mormj, 2022-01-20:
Closing due to the increase in pygccxml min requirement to 2.0

closed 2022-01-20 by mormj