gnuradio/gnuradio #3599: mpsk_stage6.grc example flowgraph output wrong

guokr321, 2020-06-23:
Hello,
I run the mpsk_stage6.grc example flowgraph of https://wiki.gnuradio.org/index.php/Guided_Tutorial_PSK_Demodulation, using gnuradio v3.7.13.5 and master version respectively, the output bit stream and original bit stream have been witten into Filesink, but they are different. How to get them synchronized?

guokr321, 2020-06-23:
Further more, using QPSK, input 1000 0x0000, 0xFFFF, 0x5555, and 0xAAAA into the Constellation Modulator, the output bit stream is the same as the input bit stream, but input random hex, the output  bit stream is different from the input bit stream, and none pattern can't be found.

duggabe, 2020-06-28:
Did you use the new flowgraph in https://github.com/gnuradio/gnuradio/blob/master/gr-digital/examples/mpsk_stage6.grc or the old one in https://github.com/gnuradio/gr-tutorial/blob/master/examples/tutorial7/mpsk_stage6.grc ?

guokr321, 2020-06-28:
both. the new mpsk_stage6.grc example flowgraph is run under master version. the old mpsk_stage6.grc example flowgraph is run under v3.7.13.5. the result is same.

duggabe, 2020-07-05:
I have found, corrected, and tested the problems with mpsk_stage6.grc
The order of blocks should be Constellation_Decoder, Differential_Decoder, and Map.
The receive Map should be [0,1,2,3].
A corrected flowgraph is in https://pastebin.com/2mRCzmPF
(The delay is 58.)
I tested with Ubuntu 18.04 and GRC 3.8.1.0; also Ubuntu 20.04 and GRC 3.9.0.0-git

I will write a PR once you have tested it.

guokr321, 2020-07-06:
> I have found, corrected, and tested the problems with mpsk_stage6.grc
> The order of blocks should be Constellation_Decoder, Differential_Decoder, and Map.
> The receive Map should be [0,1,2,3].
> A [...]

I save the link as mpsk7.grc, and
Loading: "/home/frontier/work/rcs2test/mpsk7.grc"
Error: while scanning for the next token
found character '\t' that cannot start any token
  in "/home/frontier/work/rcs2test/mpsk7.grc", line 2, column 1
>>> Failure

duggabe, 2020-07-06:
Once you go to the pastebin link, you need to click on the 'raw' icon to get the unformatted text. Alternately, you can click on the 'download' icon.

guokr321, 2020-07-07:
In the GRC 3.7.13.5, I modified mpsk_stage6.grc according your suggestion, and tested OK.

guokr321, 2020-07-08:
By the way, how to support 8PSK and 256QAM？

duggabe, 2020-07-09:
> By the way, how to support 8PSK and 256QAM？

That would be material for another tutorial...

closed 2020-07-09 by mormj

andremartins4, 2021-01-06:
The file is no longer available, is there a way to find? Thank you

duggabe, 2021-01-06:
The pastebin file mentioned above was a temporary measure until the revised file was merged into github.
The tutorial https://wiki.gnuradio.org/index.php/Guided_Tutorial_PSK_Demodulation#Decoding section has the correct link to the file:
https://raw.githubusercontent.com/gnuradio/gnuradio/master/gr-digital/examples/mpsk_stage6.grc