gnuradio/gnuradio #3604: module 'gnuradio.filter' has no attribute 'rational_resampler_fcc'

duggabe, 2020-06-24:
I am creating an example for USRP Sink and have this flowgraph: https://wiki.gnuradio.org/index.php/File:USRP_AM_xmt_fg.png
I get the following error:
```
Traceback (most recent call last):
  File "/home/barry/GRdev/USRP_AM_xmt.py", line 229, in <module>
    main()
  File "/home/barry/GRdev/USRP_AM_xmt.py", line 205, in main
    tb = top_block_cls()
  File "/home/barry/GRdev/USRP_AM_xmt.py", line 104, in __init__
    self.rational_resampler_xxx_0 = filter.rational_resampler_fcc(
AttributeError: module 'gnuradio.filter' has no attribute 'rational_resampler_fcc'
```
the resampler has 'Float->Complex(Complex taps)'
Ubuntu 19.10
GRC 3.8.1.0

willcode, 2020-06-24:
rational_resampler_fcc is missing from the python wrapper. It is in the yml, and is instantiated in the C++ code

duggabe, 2020-08-17:
This is fixed and tested in 3.9.0.0-git

closed 2020-08-17 by duggabe