gnuradio/gnuradio #3612: qtgui: intermittent error message when using QT GUI Frequncy Sink

drmpeg, 2020-06-26:
Here's the error message. It usually fails, but occasionally it works okay. This is on 32-bit armv7.
```
OverflowError: Python int too large to convert to C long

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ubuntu/xfer/USRP_FM_stereo.py", line 362, in <module>
    main()
  File "/home/ubuntu/xfer/USRP_FM_stereo.py", line 338, in main
    tb = top_block_cls()
  File "/home/ubuntu/xfer/USRP_FM_stereo.py", line 167, in __init__
    self._qtgui_freq_sink_x_0_win = sip.wrapinstance(self.qtgui_freq_sink_x_0.pyqwidget(), Qt.QWidget)
SystemError: <built-in method pyqwidget of PyCapsule object at 0xb0df3398> returned a result with an error set
```

drmpeg, 2020-06-28:
@mormj The fix for this issue is to change PyLong_AsLong() to PyLong_AsLongLong() in freq_sink_c_python.cc.

I'll create a pull request to address this issue and change all the other instances of PyLong_AsLong().

drmpeg, 2020-06-29:
https://github.com/gnuradio/gnuradio/pull/3613

mormj, 2020-06-29:
Closed by #3613

closed 2020-06-29 by mormj