gnuradio/gnuradio #3618: Embedded python in GRC Hier Block does not work

reald, 2020-07-02:
Affected version: 3.8.1.0 (Ubuntu 20.04)

# Steps to reproduce:

1. Start gnuradio-companion
2. Create HIER flow graph with some embedded python block:
[image]
3. Execute flow graph to install Hier Block
 
# Expected behaviour:
Hier Block is installed and available in GRC

# Actual behaviour:
Error:

```
Generating: '/home/user/.grc_gnuradio/testhier.py'

Executing: /usr/bin/python3 -u /home/user/.grc_gnuradio/testhier.py

Traceback (most recent call last):
  File "/home/user/.grc_gnuradio/testhier.py", line 17, in <module>
    import epy_block_0
ModuleNotFoundError: No module named 'epy_block_0'

>>> Done (return code 1)
```

reald, 2020-07-03:
Workround: Copy epy_block_0.py from project folder to ~/.grc_gnuradio and restart gnuradio-companion. 

But this does not seem the solution because the file id seems to be always "epy_block_0.py". This would only allow 1 embedded python here block installation per system. Or is there any possibility to change the block id name?

duggabe, 2020-07-04:
If you have multiple flowgraphs in the same directory and more than one has an Embedded Python block, the system will create different names for each one. For example, look at https://github.com/duggabe/gr-morse-code-gen where it created epy_block_0_0.py

I have used 3.8 with Ubuntu 18.04, 18.10, and 19.10 with no problems. I am using GRC 3.9.0.0-git on 20.04 and haven't had any problems with existing epy blocks.

reald, 2020-07-04:
@duggabe : True, and the third will be named "epy_block_0_0_0.py". This works with normal flow graphs in the same directory. In my example I try to install hier blocks. The flow graph will be installed in ~/.grc_gnuradio becomes available as block for usage in other flow graphs. (See Options -> Generate Options -> "Hier Block")

For me this installation procedure fails because no epy_block_*.py files are copied to ~/.grc_gnuradio. As a workaround I suggested in my comment to copy the epy_block_*.py by hand but this would overwrite already existing files. It might be possible to rename a file and patch the python flowgraph file by hand, but this is not comfortable.

One solution could be to store the generated embedded python files with a naming scheme like <flowgraphname>_epy_[n].py. Then it would also be more clear from which flowgraph the files were generated and cleaning up files would be more easy.
At least the hier block installer should work automatically and not collide with other epy files.

skoslowski, 2020-07-20:
Thanks for reporting these issues. 
I have been working on fixing them, see #3645.

mbr0wn, 2021-05-14:
Was fixed by @skoslowski 's PR.

closed 2021-05-14 by mbr0wn