gnuradio/gnuradio #3631: gnuradio does not follow XDG Base Directory Specification and clutters people's home directories

Avamander, 2020-07-09:
For some reason gnuradio creates a folder `~/.gnuradio` and uses it for configuration. That's not in-line with [XDG Base Directory Specification](https://wiki.debian.org/XDGBaseDirectorySpecification) and clutters people's home directories.

gnuradio should follow XDG base directory specification and use `~/.config/gnuradio` for configuration files

DC7IA, 2021-11-10:
> gnuradio should follow XDG base directory specification and use `~/.config/gnuradio` for configuration files

gnuradio should take the path from the appropriate variable

DC7IA, 2021-11-10:
Here's the link to the spec, as you only linked to the Debian wiki: https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html

ostrich, 2023-08-28:
Thanks @willcode. `gnuradio-config-info --userprefsdir` shows the value of $GR_PREFS_PATH but GNU Radio Companion is still creating grc.conf in ~/.gnuradio. Does GRC have support for a different environment variable?

willcode, 2023-08-28:
GRC_PREFS_PATH. Please use chat or the mailing list for any further help. This github issue is about a specific feature request.

marcusmueller, 2024-01-07:
still desirable, still sensible, still something worth doing in both 3.x and 4.x development

closed 2024-02-29 by willcode