gnuradio/gnuradio #3635: Weird toolbar scaling on GRC

ruilvo, 2020-07-10:
The toolbar on GRC is too large, and icons are inconsistent in size.

OS information: Manjaro KDE 20.0.3 fresh install, persistent after full update.
GNURadio installed from pacman, reproducible also with gnuradio-git from AUR.

What I did also:
This is a laptop with a small display, so I tried with 125% scaling on KDE, and text DPI only scaling. Neither affected the bug. 
Messing with DPI_SCALING in grc/gui/Constants.py didn't seem to have any effect. 

Screenshots:
Notice how the "printer" icon is larger than the others. The bar can't fit all icons.
[image]
After reinstalled the OS the icons that got large changed seemingly arbitrary. 
[image]

tanaxer01, 2021-08-25:
I was having the same issue, but after changing the icons to the ones used in GNOME the problem stopped

Before:
[image]

After:
[image]

After tinkering with different icon sets , I only saw this issue with the Breath2 icons

Hope it helps!

marcusmueller, 2023-09-22:
I'm going to close this issue as solved – this seems to be a bug in the breath2 gnome icon theme.

closed 2023-09-22 by marcusmueller