gnuradio/gnuradio #3637: Upstreaming some blocks from gr-satellites

daniestevez, 2020-07-11:
@bistromath has [suggested over on Twitter](https://twitter.com/bistromath/status/1281941914011435011) that it might be good to upstream some of the blocks used in [gr-satellites](https://github.com/daniestevez/gr-satellites). I think he's right. I try to do things with upstream blocks as much as possible, but there are some blocks I've needed to write and which might be useful in a more general context. I'm opening this issue mainly as a self-reminder, and also for additional discussion with the community about the best way to proceed.

This is a rough list of the blocks that off the top of my head I think it could be interesting to upstream.

### HDLC framer/deframer

Upstream already has their own blocks. The difference between the upstream and gr-satellites blocks is that the gr-satellites blocks are done in Python, the framer allows to specify the length of the preamble and postamble (upstream uses only 1 repetition of the `0x7e` flag), the deframer allows to skip the CRC-16 check (useful for debugging), and the framer outputs PDUs instead of bytes (might be interesting for burst transmissions).

I think that C++ is the way to go, for performance, so rather than upstreaming the gr-satellites blocks, it would be good to incorporate the additional options into the upstream blocks.

The point I'm not sure how to go about is the difference between bytes output and PDU output for the framer. Maybe it's best to support both options?

### NRZI

HDLC often goes with NRZI, which is some kind of differential encoding plus an inversion. This can be achieved with upstream blocks (Differential encoder and Not), but it is nice to have a more compact solution, as gr-satellite does (one NRZI encoder block, one NRZI decoder block) These would be easy to upstream. They're now written in C++ and have some Python QA code.

### Reed-Solomon

The "most general" Reed-Solomon block in gr-satellites is the "Reed Solomon Decoder (libfec)" block. This is basically a wrapper for the `init_rs_char()` and `decode_rs_char()` functions in Phil Karn's libfec. I'm not sure how this would fit into FECAPI. Maybe it's best to do FECAPI blocks for Reed-Solomon (which I think don't exist), since I'm quite happy using FECAPI for Viterbi decoding.

### Matrix interleaving/deinterleaving

The prototypical algorithm where data is written onto a matrix by columns and read by rows. I don't think it's possible to do this easily with upstream blocks. gr-satellites has a fairly general block for matrix deinterleaving (it allows to define the matrix size and also to crop the output). However, it is only implemented for floats. This should be implemented as a template to allow all the data types.

### Generating PDUs by finding a syncword

I think this is a prototypical example, especially for simple packet protocols. We've marked our stream with tags corresponding to the syncword locations by using Correlate Access Code - Tag Stream. Now we want to extract PDUs of a predetermined length starting at each of these tags locations. We want to do this even if such PDUs would "overlap", to protect ourselves from syncword misdetections. So some of the samples might end into several PDUs.

There is a Python block in gr-satellites that does that, when used in combination with Tagged Stream to PDU. The block is called `fixedlen_tagger`.

But there might be a way to do this very same thing using upstream blocks. I'm not so familiar with the Protocol parser block, for instance, so I don't know if it could be used for this.

drmpeg, 2020-07-11:
Reed-Solomon already exists in gr-fec. Here's an example of usage:

https://github.com/gnuradio/gnuradio/blob/master/gr-dtv/lib/dvbt/dvbt_reed_solomon_enc_impl.cc

https://github.com/gnuradio/gnuradio/blob/master/gr-dtv/lib/dvbt/dvbt_reed_solomon_dec_impl.cc

daniestevez, 2020-07-12:
Thanks Ron! I didn't know that.

I have tried to make gr-satellites use the Reed-Solomon functions from gr-fec and drop the dependence on libfec. Unfortunately, my first approach doesn't work. It seems that the functions `encode_rs_8()` and `decode_rs_8()` are not exported in `libgnuradio-fec.so`, even though they are [declared in the header `<gnuradio/fec/rs.h>`](https://github.com/gnuradio/gnuradio/blob/5a7a46e27e5fa1effb0ead8914955b2ac8a2c180/gr-fec/include/gnuradio/fec/rs.h#L30):

```
$ objdump -T /usr/lib64/libgnuradio-fec.so | grep "[en|de]code_rs"
000000000003a570 g    DF .text	0000000000000119  Base        encode_rs_char
000000000003a690 g    DF .text	0000000000000a9b  Base        decode_rs_char
```

The functions `encode_rs_8()` and `decode_rs_8()` are a special case of the general `encode_rs_char()` and `decode_rs_char()` (for a particular Reed-Solomon code). However, the Reed-Solomon code they implement is so common that they are handy to have.

Some additional "gotchas" I've found:

* `encode_rs_ccsds()` and `decode_rs_ccsds()` are not declared in `<gnuradio/fec/rs.h>`, even though [they are implemented](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/lib/reed-solomon/encode_rs_ccsds.c).

* The [libfec version](https://github.com/quiet/libfec/) I was using has an extra `padding` parameter in the encode and decode functions. The implementation in GNU Radio doesn't have this parameter. This can be handled by zero padding the beginning of the message externally, as [the DVB-T RS blocks do](https://github.com/gnuradio/gnuradio/blob/5a7a46e27e5fa1effb0ead8914955b2ac8a2c180/gr-dtv/lib/dvbt/dvbt_reed_solomon_enc_impl.cc#L76). Alternatively (this is mainly a note to my future self), I believe that due to the cyclic nature of RS codes the zero padding can be done at the end of the message instead. This can be useful to avoid copying in some cases, for instance, if the message already has a header at the beginning.

With this in mind, I think we could do the following changes to gr-fec:

1. Export `encode_rs_8()` and `decode_rs_8()` in `libgnuradio-fec.so`. Export `encode_rs_ccsds()` and `decode_rs_ccsds()` in `libgnuradio-fec.so` and add their declarations to `<gnuradio/fec/rs.h>`.

2. I'm not sure on how to go about the padding. If you look at how it's handled in libfec using the `padding` parameter, you'll see that it's just used to modify indices in a [few](https://github.com/quiet/libfec/blob/master/decode_rs.h#L85) [places](https://github.com/quiet/libfec/blob/master/decode_rs.h#L289), without actually padding anything. This seems more efficient than actually having to pad with zeros. So my suggestion would be to add the `padding` parameter to gr-fec, but things that use it (such as gr-dtv) would also need to be changed.

If we agree to these changes, I could try to do them and submit one or two PRs, for instance starting with 1., which doesn't lend to much discussion and seems quite easy to do (but first I'll need to get my GNU Radio master branch dev environment going).

Now, going back to the original topic of whether adding a generic block for Reed-Solomon to upstream would be good. Looking at the DVB-T Reed-Solomon blocks, I see there isn't anything special about DVB-T in them. They just implement encoding/decoding of a generic RS code using `encode_rs_char()`. (Well, I guess that the default parameters of the GRC block correspond to the code used in DVB-T). So maybe it would be least confusing for users if these blocks weren't classified under DVB-T in GRC (I see also RS blocks for ATSC and ITU-T J.83B which have the RS code definition baked in). In fact one could argue that the generic RS blocks could be moved from gr-dtv to gr-fec, but this is mostly transparent to the user.

Now, the thing about the DVB-T Reed-Solomon blocks that doesn't fit my use case so well is that they use a vector stream of chars. This makes a lot of sense for DVB-T, but for packetized data I prefer PDUs (which is what the blocks in gr-satellites do). With the [FECAPI deployments](https://www.gnuradio.org/doc/doxygen/page_fec.html#fec_using) I can easily switch between a char stream (using FEC Encoder) to PDUs (using FEC Async Encoder). However, there is no option to use vector streams, and I don't think this is so trivial, since GRC would need to know the block sizes of the underlying code, but maybe it's doable.

So anyway, even though in general I think that gr-fec would lend to some work in trying to have the same kind of interface and functionality for all the codes, this is probably a major task. For something more simple regarding just Reed-Solomon, maybe a good solution is to have a GRC `.yml` file for the current DVB-T RS blocks that classifies them under "Error Coding" and includes default options for the CCSDS code (which is the most popular), and new blocks called "Reed-Solomon Async (En|De)coder" that do the same but with PDUs.

A final note (just to make things more complicated): the `encode_rs_ccsds()` codes (CCSDS code with dual basis) are also rather popular and I think these can't be used with the `decode_rs_char()` generic framework. These involve [switch of basis](https://github.com/gnuradio/gnuradio/blob/5a7a46e27e5fa1effb0ead8914955b2ac8a2c180/gr-fec/lib/reed-solomon/encode_rs_ccsds.c#L16) to and from the conventional basis. So maybe specific new blocks are needed for this.

daniestevez, 2020-07-19:
I've just sent a PR that achieves

> Export encode_rs_8() and decode_rs_8() in libgnuradio-fec.so. Export encode_rs_ccsds() and decode_rs_ccsds() in libgnuradio-fec.so and add their declarations to <gnuradio/fec/rs.h>.

drmpeg, 2020-07-19:
I wouldn't worry about trying to make a generic Reed-Solomon block that satisfies gr-dtv functionality. I'd say a generic block that meets just your needs is fine. I don't think bloat is really much of a concern in GNU Radio.

jdemel, 2020-07-22:
@daniestevez should your matrix interleaver be adjustable on runtime? I added a similar interleaver to [gr-symbolmapping](https://github.com/ant-uni-bremen/gr-symbolmapping). It could be adjustable on runtime but only with a new vector of interleaver positions. It is templatized though. I just generate the positions where all the input values should go and loop over this vector.

daniestevez, 2020-07-22:
@jdemel in gr-satellites the matrix interleaver is not adjustable at runtime. For this use case it's enough to define the parameters when the block is instanced. I don't have a use case in mind where parameters would need to be changed on the fly. Most protocols I know define a fixed interleaving scheme (or perhaps several if there are several predefined frame sizes). I could imagine a case when the frame has a variable size `n*k` (for `k` fixed) and the data should be written in an `n x k` matrix by rows and read by columns, but I'm not sure greater flexibility is needed in most cases (but would be fantastic to have, just in case).

jdemel, 2020-07-22:
So it would be enough to just use a vector with interleaved positions in the constructor and then just loop over it? A matrix interleaver could then be initialized with
```python
oot.interleaver(np.reshape(np.arange(vec_len), (n_rows, n_cols)).T.flatten())
```
Would that work?

daniestevez, 2020-07-22:
I guess that would work, and also the special cases also allowed by the gr-satellites block which do things like starting to read from the matrix with an offset.

If I understood correctly, your block defines explicitly an permutation from the input vector positions to the output vector positions, so I can't think of an interleaving scheme that can't be obtained with it.

jdemel, 2020-07-29:
Your interleaver uses message passing. The one I implemented targets the streaming interface so far. I guess we could reuse most of the code for both interfaces. Then we might just add different GRC files to specialize like the fast random source.

daniestevez, 2020-07-29:
In fact both streaming and PDU would make sense depending on the intended application. The same happens for Reed-Solomon or other block FEC.

mbr0wn, 2021-05-14:
@daniestevez what's the status of this issue? Are you intending to upstream all five blocks (or have so already)? I just need to know what the condition is for us being able to close this. Thanks for upstreaming those blocks.

daniestevez, 2021-05-15:
Hi Martin, unfortunately I have progressed very little with all this. I should really push myself into finding some time to advance with this. Regarding each of the blocks, I think that the order by difficulty / time needed is:

1. NRZI. This is easy. Just copy the code over in a PR. Let me try to tackle this perhaps next weekend or so.
2. Reed-Solomon. I think that the current Reed-Solomon blocks in gr-satellites would be a good addition to GNU Radio. For this I first wanted to refactor the blocks in gr-satellites to use the libfec functions inside GNU Radio (currently gr-satellites also embeds a portion of libfec for this), and then move the blocks in tree once I've checked they work (there are small API differences between both libfec's). However, #4454 is preventing me from doing it. I could circumvent this by directly moving the blocks in tree, but I think it's important to fix #4454. I would need some help with that, since it has to do with the way that library symbol are exported on Windows, and I don't know much about that.
3. HDLC deframer. Probably the way to go here is to implement some of the extra features that gr-satellites has on the in-tree block.
4. Matrix interleaving. There was some discussion about whether a more general block that allows any kind of interleaving pattern is more useful (see https://github.com/gnuradio/gnuradio/issues/3637#issuecomment-662491711). I agree with this, so it's probably easier move in tree the block from gr-symbolmapping that Johannes mentioned.
5. PDUs and syncwords. In fact I did a naïve C++ re-implementation of my Python block, to see how the performance was. Surprisingly the performance was much worse. I think that I should re-think the implementation approach of this block, since currently it uses a Python deque to hold old samples rather than taking advantage of the history. There is a similar block in [gr-pdu_utils](https://github.com/sandialabs/gr-pdu_utils), but it doesn't handle the case of overlapping data (when a new tag happens before the previous tag + the frame length).

So I think it is realistic for me to try to do 1-3 over the next several weeks and leave out 4-5, since the approach for those is not so clear.

jaredd, 2021-07-13:
FYI, I noticed this conversation and submitted a matrix interleaver implementation I had as a hier block using the deinterleave and interleave functions. PR #4861. Seems to work, but might not satisfy all the requirements people have here (like offset reading).