gnuradio/gnuradio #3638: Wrong C++ code generated for low and high pass filters

sashko, 2020-07-11:
GNU Radio version 3.8.1.0.

1. Create simple flow graph: Null source -> High Pass Filter -> Null Sink. Generate C++ code.
2. Compile code.

Result:

```
[ 50%] Building CXX object CMakeFiles/test1.dir/test1.cpp.o
In file included from /home/sashko/test1/test1.cpp:9:
/home/sashko/test1/test1.hpp:30:21: error: ‘sptr’ in ‘class gr::filter::firdes’ does not name a type
   30 |     filter::firdes::sptr high_pass_filter_0;
      |                     ^~~~
/home/sashko/test1/test1.cpp: In constructor ‘test1::test1()’:
/home/sashko/test1/test1.cpp:22:15: error: ‘class test1’ has no member named ‘high_pass_filter_0’
   22 |         this->high_pass_filter_0 = filter::fir_filter_ccf::make(
      |               ^~~~~~~~~~~~~~~~~~
/home/sashko/test1/test1.cpp:22:44: error: ‘gr::filter::fir_filter_ccf’ has not been declared
   22 |         this->high_pass_filter_0 = filter::fir_filter_ccf::make(
      |                                            ^~~~~~~~~~~~~~
/home/sashko/test1/test1.cpp:24:13: error: ‘firdes’ was not declared in this scope; did you mean ‘gr::filter::firdes’?
   24 |             firdes.high_pass(
      |             ^~~~~~
      |             gr::filter::firdes
In file included from /home/sashko/test1/test1.hpp:17,
                 from /home/sashko/test1/test1.cpp:9:
/usr/include/gnuradio/filter/firdes.h:40:18: note: ‘gr::filter::firdes’ declared here
   40 | class FILTER_API firdes
      |                  ^~~~~~
/home/sashko/test1/test1.cpp:27:17: error: expected primary-expression before ‘,’ token
   27 |                 ,
      |                 ^
/home/sashko/test1/test1.cpp:28:17: error: expected primary-expression before ‘,’ token
   28 |                 ,
      |                 ^
/home/sashko/test1/test1.cpp:40:73: error: ‘class test1’ has no member named ‘high_pass_filter_0’
   40 |     this->tb->hier_block2::connect(this->blocks_null_source_0, 0, this->high_pass_filter_0, 0);
      |                                                                         ^~~~~~~~~~~~~~~~~~
/home/sashko/test1/test1.cpp:41:42: error: ‘class test1’ has no member named ‘high_pass_filter_0’
   41 |     this->tb->hier_block2::connect(this->high_pass_filter_0, 0, this->blocks_null_sink_0, 0);
      |                                          ^~~~~~~~~~~~~~~~~~
/home/sashko/test1/test1.cpp: In member function ‘void test1::set_samp_rate(double)’:
/home/sashko/test1/test1.cpp:54:11: error: ‘class test1’ has no member named ‘high_pass_filter_0’
   54 |     this->high_pass_filter_0->set_taps(firdes::high_pass(1, this->samp_rate, , , firdes.WIN_HAMMING, 6.76));
      |           ^~~~~~~~~~~~~~~~~~
/home/sashko/test1/test1.cpp:54:40: error: ‘firdes’ has not been declared
   54 |     this->high_pass_filter_0->set_taps(firdes::high_pass(1, this->samp_rate, , , firdes.WIN_HAMMING, 6.76));
      |                                        ^~~~~~
/home/sashko/test1/test1.cpp:54:78: error: expected primary-expression before ‘,’ token
   54 |     this->high_pass_filter_0->set_taps(firdes::high_pass(1, this->samp_rate, , , firdes.WIN_HAMMING, 6.76));
      |                                                                              ^
/home/sashko/test1/test1.cpp:54:80: error: expected primary-expression before ‘,’ token
   54 |     this->high_pass_filter_0->set_taps(firdes::high_pass(1, this->samp_rate, , , firdes.WIN_HAMMING, 6.76));
      |                                                                                ^
/home/sashko/test1/test1.cpp:54:82: error: ‘firdes’ was not declared in this scope; did you mean ‘gr::filter::firdes’?
   54 |     this->high_pass_filter_0->set_taps(firdes::high_pass(1, this->samp_rate, , , firdes.WIN_HAMMING, 6.76));
      |                                                                                  ^~~~~~
      |                                                                                  gr::filter::firdes
In file included from /home/sashko/test1/test1.hpp:17,
                 from /home/sashko/test1/test1.cpp:9:
/usr/include/gnuradio/filter/firdes.h:40:18: note: ‘gr::filter::firdes’ declared here
   40 | class FILTER_API firdes
      |                  ^~~~~~
make[2]: *** [CMakeFiles/test1.dir/build.make:63: CMakeFiles/test1.dir/test1.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/test1.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

haakov, 2020-07-17:
Thanks, I'll take care of it

Cyclic3, 2021-05-17:
Hi, has there been any progress on this? If not, how hard would it be for someone new to the project (but not to C++) to implement it?

marcusmueller, 2024-01-22:
Persists, Demo FG: https://gist.githubusercontent.com/marcusmueller/e6931b42db953a5bc12fcffadb9a125d/raw/0a34837d23659a58d5ad2377521a3ed06bd63868/repro3638.grc