gnuradio/gnuradio #3639: qtgui: time raster sink -- incompatible constructor arguments

citizenrich, 2020-07-11:
I'm seeing an issue in current 3.9 releases-master PPA with QT GUI Time Raster. This can be reproduced using the examples.
```
richard@hp:~/gnuradio/gr-qtgui/examples$ python3 ./pyqt_time_raster_b.py 
Traceback (most recent call last):
  File "./pyqt_time_raster_b.py", line 73, in <module>
    tb = my_top_block();
  File "./pyqt_time_raster_b.py", line 56, in __init__
    self.snk1 = qtgui.time_raster_sink_b(fs, nrows, ncols, [], [],
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gnuradio.qtgui.qtgui_python.time_raster_sink_b(samp_rate: float, rows: float, cols: float, mult: List[float], offset: List[float], name: str, nconnections: int = 1, parent: gnuradio.qtgui.qtgui_python.QWidget = 0)

Invoked with: 200, 100, 100.25, [], [], 'Time Raster Example', 2
richard@hp:~/gnuradio/gr-qtgui/examples$ python3 ./pyqt_time_raster_f.py 
Traceback (most recent call last):
  File "./pyqt_time_raster_f.py", line 72, in <module>
    tb = my_top_block();
  File "./pyqt_time_raster_f.py", line 55, in __init__
    self.snk1 = qtgui.time_raster_sink_f(fs, nrows, ncols, [], [],
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gnuradio.qtgui.qtgui_python.time_raster_sink_f(samp_rate: float, rows: float, cols: float, mult: List[float], offset: List[float], name: str, nconnections: int = 1, parent: gnuradio.qtgui.qtgui_python.QWidget = 0)

Invoked with: 200, 100, 100.25, [], [], 'Float Time Raster Example', 2

```

drmpeg, 2020-07-11:
Verified here on source build of latest master. I've also seen this intermittently on an OOT flow graph, but I can't reproduce it yet.

drmpeg, 2020-07-17:
I've been able to track this down. The line:
```
        self.snk1 = qtgui.time_raster_sink_b(fs, nrows, ncols, [], [],
                                             "Time Raster Example", 2)
```
is missing a parameter. It should be:
```
        self.snk1 = qtgui.time_raster_sink_b(fs, nrows, ncols, [], [],
                                             "Time Raster Example", 2, None)
```
The problem also exists in qtgui_time_raster_x.block.yml.

drmpeg, 2020-07-19:
PR submitted. https://github.com/gnuradio/gnuradio/pull/3658

drmpeg, 2020-07-26:
Merged. https://github.com/gnuradio/gnuradio/commit/c2f42238032e2315371aaec8a64687d99028ccd2

closed 2020-07-26 by drmpeg